OctoPrint/OctoPrint #2014: Filament volume is not calculated

Mavyre, 2017-07-18:
#### What were you doing?
Printing anything with Octoprint

#### What did you expect to happen?
Filament usage in volume is calculated and displayed

#### What happened instead?
Only the filament length usage is displayed (see screenshot)

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint
Raspberry Pi 3 using OctoPi 0.13.0

#### Printer model & used firmware incl. version
Anet A8, default firmware

#### Browser and Version of Browser, Operating System running Browser
Windows 10, Chrome 59.0.3071.115

#### Link to octoprint.log
https://pastebin.com/T1UNbtGK

#### Link to contents of terminal tab or serial.log
Don't have it and cannot launch a print to test for now :(

#### Link to contents of Javascript console in the browser
[image]

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2017-07-18:
Please provide an example GCODE file with which to reproduce.

As a general rule of thumb, volume can only be calculated if the filament diameter is known. So if that can't be parsed from the (non-standardized) comments left in the GCODE by slicers, no way to calculate it and hence it stays empty.

Mavyre, 2017-07-18:
I'm using Cura 2.6, here is an example of the first few lines of a Gcode file:
https://pastebin.com/2pGEprbj

Could I add the filament size as a comment in the starting Gcode section?

foosel, 2017-07-18:
Please provide a _full_ file, it might be that what we are looking for is at the end or somewhere in the middle ;)

OctoPrint currently supports comments generated by Cura 15.x, Slic3r (probably also the Prusa fork) and Simplify3D. Cura 15.x encoded all slicing settings in a long base64 string, so that would be a bit fiddly to manually add as a comment. Slic3r and S3D are easier:

    ;filament_diameter=1.75 ; Slic3r format
    
    ;filamentDiameter,1.75  ; S3D format

My guess is the format changed between Cura 15 and the newer Cura 2.x, so a new parsing routine needs to be added. To do that I need a valid sample.

Mavyre, 2017-07-18:
Here you go:
[AA_Anet_A8_Chain_Bed_Protector_v2.gcode.txt](https://github.com/foosel/OctoPrint/files/1155451/AA_Anet_A8_Chain_Bed_Protector_v2.gcode.txt)

Pastebin doesn't support it as it's too big.

foosel, 2017-07-18:
Ok... that sadly looks like there's nothing that can be done from OctoPrint. I don't see any kind of information on slicing parameters incl. the filament diameter in there. Without that all I have is the length of extruded filament. No way to calculate the volume just with that, one dimension is missing ;)

Too bad they left out that data from their generated output now, it was useful.

foosel, 2017-07-18:
Closing this since the information needed to solve the issue is simply not available, and there's nothing to be done in OctoPrint about it. Unless Cura 2.x adds back some way to get the filament diameter from sliced GCODE files, there's no way to support volume information for such files. Sorry.

closed 2017-07-18 by foosel

Mavyre, 2017-07-18:
Switching to Slic3r then :)

Thank you a lot for your help

leppa, 2018-03-05:
Didn't want to open a new issue just for that, but I have a suggestion: why not adding a way to set the default filament diameter (e.g., to a printer profile)?
This way, if there's no filament diameter information in GCODE, a default value can be used.

tobashadow, 2018-06-25:
Adding to this one since I am having the same issue, in the latest Cura 3.3.1 at the very bottom of the GCODE is a section called settings_3 and in that is the following line 

nmaterial_diameter = 1.75