OctoPrint/OctoPrint #2015: Username capital characters on login

E000001, 2017-07-18:
Hello,
I am not sure if it is an issue. But I spent a lot of time to login with correct username and password to Octoprint web interface. 

I created new user named Pi in initial settings (Android keyboard makes automatically first letter capital). When I want to try to login first time. I was typing pi with lower P. :/ And Octoprint wrote to me "Logon Failed" :/

So, It is possible to disable this Match casing?

Thank you in advace

E

foosel, 2017-07-21:
OctoPrint already sets `autocapitalize="none"` on the user name field for that very reason, which all mobile keyboards I have access to appear to honor (various Android & iOS).

Doing an case insensitive match on the user name is something I do not feel comfortable about (it is a very small security advantage having to also match the case of the username but it *is* a security advantage), and doing so only to get around software keyboards that do not honor hints on input fields that disable auto capitalization feels especially meh.

Sorry, but this is not going to be changed for the stock user manager in the foreseeable future. Stuff is modular though, and starting with 1.3.5 it's even more easy to override the used user manager through a new plugin hook, so it would be possible to write a plugin that makes user names be case insensitive. Marking this as "plugin only".

6cUbi57z, 2019-06-17:
The autocapitalise attribute doesn't seem to be present in the latest version (1.3.11, OctoPi 0.16.0 Browsing using Firefox 67.0.1 on Windows 10 and Firefox on Android). Looks like this is due to the introduction of the force login plugin.

I've ran into this in a number of pieces of software and always find it annoying because it usually breaks login on mobile. I don't really see the security argument, I can see it being a big pain for support. The only account I could see it adding any security benefit to would be the typical admin account names (An attacker might try Admin, admin, Administrator, administrator, etc.) but these are typically considered bad practice now.

The workaround I usually go with is just to create two user accounts. One with the first character capitialised and the other not.

The bigger security concern for would be that autocomplete seems to be turned on for the username field. All you have to do to get a list of all users who have logged on from your machine is click in the text field.