OctoPrint/OctoPrint #2017: gcode viewer does not show brim

Andrei-Pozolotin, 2017-07-19:
var DISPLAY_VERSION = "1.3.4 (master branch)";

gcode viewer does not show brim: as follows:

1) the attached gcode file
[export - corner-2.gcode.zip](https://github.com/foosel/OctoPrint/files/1159436/export.-.corner-2.gcode.zip)

2) as rendered in octoprint
[image]

3) and as rendered in on-line viewver http://gcode.ws/
[image]

the difference is: the brim is missing in octoprint. perhaps this bug should be a feature?

foosel, 2017-07-24:
That one seriously puzzled me at first. Until I prepared a minimal example by stripping away most of the file and then noticed that if it just had the brim bits in there it displayed just fine, and once the actual model started that apparently overwrote some of the information. 

Caused by a wrong comparison basically, fixed by the above commit on `maintenance`, soon `devel` and to be pushed out with the 1.3.5 release.

foosel, 2017-07-24:
PS: Please always use and fill out the complete ticket template, it helps in processing. This time all information I needed was there since I could immediately reproduce via that included (Thanks!) GCODE file, but if that hadn't been the case the lack of the other information would have prevented a speedy resolution :)

commit 0456896d (2017-07-24): GCODE Viewer: Fix issue with lookup of layer index 0

We were just testing if z_heights[z] was truthy. If the layer index was
set but 0 that wasn't properly detected and let to overwriting our
layer information. Instead check explicitly against undefined.

Closes #2017
closed 2017-10-16 by foosel