OctoPrint/OctoPrint #2018: [Request] Disable ability to download GCode for Guest ACL

WhatsADSM, 2017-07-19:
As it sits today a guest user is allowed to perform a few actions that expose the users IP.

Namely they are allowed to download the GCode. While not the most straight forware thing from the GCode you can pretty much reconstruct a model. If you were to have the same printer you could potentially just use the GCode as is.

I'm not sure what the justification for allowing the user to download the gcode was but I suggest that the feature is removed.

Other notable things, the user can perform some SD operations like releasing the card, and they can also see the terminal activity which again indirectly exposes the IP.

foosel, 2017-07-20:
Duplicate of #1110

closed 2017-07-20 by foosel

WhatsADSM, 2017-07-20:
For what its worth I disagree that my request is a duplicate. That request is asking for a more granular access control system based on roles. That is very powerful but complex to implement.

I am simply asking that for the guest user the download gcode option is removed, nothing else. Probably an order of magnitude easier to implement.

foosel, 2017-07-20:
Might be easier to implement. Also breaks some functionality however (e.g. the GCODE viewer) and some people want it to be available to guests too (e.g. for the GCODE viewer functionality). So it needs to be configurable. And it doesn't make sense adding options for this and that all over the place when a proper solution would also solve a bunch of other problems (and on top of things is already in the works by @Salandora).