OctoPrint/OctoPrint #2020: XYZ Davinci 1.0 not connecting to Octoprint

Jackpen7, 2017-07-20:
#### What were you doing?
I was trying to connect my XYZPrinting to Octoprint
1. ...
2. ...
3. ...

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?
I expected it to connect.
#### What happened instead?
It failed with error:
`There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor`
#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint
Windows 7 SP1

#### Printer model & used firmware incl. version

XYZPrinting Da Vinci 1.0 w/ Repetier 0.92

#### Browser and Version of Browser, Operating System running Browser
Chrome 59.0.3071.115 on Windows 10 Pro Creators Update

#### Link to octoprint.log
https://pastebin.com/MkQu292A
#### Link to contents of terminal tab or serial.log
https://pastebin.com/eL8YqMS6
#### Link to contents of Javascript console in the browser
https://pastebin.com/jGZ4KVkW
#### Screenshot(s)/video(s) showing the problem:
I have read the FAQ.

Jackpen7, 2017-07-20:
It somehow started working after closing and reopening OctoPrint.

foosel, 2017-07-24:
"Closing and opening" as in, in the browser? Or restarting?

In your logs there are three different kinds of connection problem.

  1. Autodetection of your serial port not working. That's fairly common because the way it's currently implemented in OctoPrint simply doesn't work with all printer types (it will become a bit better though in 1.3.5). Selecting the (probably only) available serial port manually and attempting to connect (ideally at the correct baudrate) should solve this.
  2. OctoPrint getting an "Access Denied" from the OS when attempting to open the port. That's usually caused by some other program that has grabbed the serial port already. Concurrent access doesn't work. If you had Simplify3D open in the background when that happened, that could be the reason - in my experience it loves to just grab every serial port it sees for itself.
  3. Your printer not answering to OctoPrint's handshake attempts (sending `M110` and hoping for an `ok` back) in a timely manner. If no answer comes back OctoPrint will assume your printer is not in fact a printer and be forced to give up. Sometimes this is simply caused by your printer taking a tiny bit too long to boot up after the for most models obligatory reset of its controller. In that case you might be able to mitigate this problem by enabling "Wait for `start` on connect" under Settings > Features. 

All of this doesn't sound like anything I could do anything about (that I haven't already done), so if this problem either isn't reproducible any longer or solved through and of the above suggestions, please close the ticket. Otherwise it might be possible to dig deeper, but it will most likely require some back and forth since I can't test against this specific printer model myself and have to rely on input. Thanks!

foosel, 2017-10-02:
Still now response from OP, considering this obsolete.

closed 2017-10-02 by foosel