OctoPrint/OctoPrint #2021: Grammar error

mikael-gaudet, 2017-07-20:
There is error when files are uploaded it says:

Uploaded: in a few seconds

Suggestion:

Uploaded: a few seconds ago

Salandora, 2017-07-21:
Hello,

that's not directly an error also not a grammer mistake, think about this scenario:

Time of OctoPrint Server: 02:00:00
Time of Computer: 01:59:00
Difference: 1 minute

If you now upload a file the OctoPrint Server says it got uploaded at 02:00:00 this is also what the server says to your webbrowser.
Now the file list on your webbrowser changes, your webbrowser is using the time on your computer so the difference ins 1 minute, so for the browser the upload of the file is in the future, which means the "in a few seconds" makes sence here.

Maybe you could try to compare your Server time and your Computer time and confirm this please :-)

ntoff, 2017-07-21:
I've also seen similar things when uploading to websites or posting posts, it says things like "posted in a few minutes" or that steam will tell me I've last played a particular game "tomorrow".

foosel, 2017-07-27:
As already pointed out, this is not a grammar error but simply caused by an out-of-sync clock between your browser and the backend.

closed 2017-07-27 by foosel