OctoPrint/OctoPrint #2023: [Request] Dark mode

HarlemSquirrel, 2017-07-22:
I would love a dark mode option for the appearance. I'm happy to build the feature and open a PR if the project is open to it. I'm thinking something like the new YouTube dark mode with light text on a dark background.

jneilliii, 2017-07-23:
Go into settings and change the theme to Black...

[image]

nophead, 2017-07-23:
That setting only changes the title bar for me.

HarlemSquirrel, 2017-07-23:
I'm running version 1.3.4 and that setting only changes the title bar for me as well.

ntoff, 2017-07-23:
http://plugins.octoprint.org/plugins/cyborgtheme/

jneilliii, 2017-07-23:
Oh yeah, forgot I had installed that. Install the cyborg theme plugin and then set the appearance setting to black to match.

HarlemSquirrel, 2017-07-23:
That works, thanks!

closed 2017-07-23 by HarlemSquirrel