OctoPrint/OctoPrint #2024: Printer Stop when touch HOME

Neoolog, 2017-07-22:
i got Octopi on my Raspberry pi3 and touch screen, so if i accendently touch the HOME button on my touch screen   in the Octopi Control Tab  ( beacause the screen are in off mode ) , the printer stop printing without confirmation

Version 1.3.4 (master branch) with smoothieware

Neoolog, 2017-07-22:
Is ok...i was think i touch the home button. but is the homemade command i do with G28 inside, i will put confirmation in my command....sorry...

foosel, 2017-07-24:
Closing then.

Did you maybe ask this also for the latest OctoPrint On Air episode that aired the day before yesterday? Because I answered that there too. Recording is available at the former live link, public one will be made available once I get around to do post processing.

closed 2017-07-24 by foosel