OctoPrint/OctoPrint #2025: [Request] Allow octoprint to just quit as an option for restart

lf-, 2017-07-23:
#### What were you doing?
Trying to set up the restart command. I set it to `kill -SIGINT $PPID`, but that didn't work to restart the server. The reason I'm trying to do this is to allow my service manager (systemd) to restart octoprint for me. Later, I tried `sh -c 'kill -SIGINT $PPID'` with a similar lack of luck. 

It appears that it is necessary for OctoPrint to implement some way of quitting itself for the restart option as my attempts to do it externally have failed.

#### What did you expect to happen?
It should have restarted.

#### What happened instead?
No processes were harmed.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Branch & Commit or Version of OctoPrint
fe481e12b3c50a2def3b41515200e13a8c6a5c72 (master)

#### Operating System running OctoPrint
debian unstable (`Debian GNU/Linux buster/sid`)

#### Printer model & used firmware incl. version
N/A

#### Browser and Version of Browser, Operating System running Browser
Chromium 59 on Arch Linux

#### Link to octoprint.log
https://gist.github.com/lf-/dce3d604c29405b8112f0c108f8617b1

foosel, 2017-07-24:
The command failed silently (I have to execute those "server will be gone right now" commands asynchronously for technical reasons and hence can't log the output or wait for the return code). When I tried replicating just via `subprocess.Popen` I saw why:
 
```
>>> import subprocess
>>> p = subprocess.Popen("kill -SIGINT $PPID", shell=True)
/bin/sh: 1: kill: Illegal option -S
```

So, `/bin/sh` got invoked and didn't like the `-SIGINT` parameter. Turned out that when invoked like that the `kill` builtin of `sh` is used, and that doesn't know the sigspec `SIGINT` (you can check that yourself via `sh -c 'kill -l'`). It does know `INT` though, and the signum `2` works as well.

So, using either `kill -INT $PPID` or `kill -2 $PPID` should get you going there. Alternatively use `/bin/kill -SIGINT $PPID`, that works around the builtin.

Shuts down the server fine for me at least. Please report back if that solves the problem for you too :)

lf-, 2017-07-24:
Oh, dammit! It works in zsh and not in sh which I think is dash. Thank you.
Builtins should be illegal.

foosel, 2017-07-24:
Thanks to the above commit it will have better logging as long as the logging subsystem is still available (which it should be unless the command succeeds in which case there's no reason to debug). And that should cover all relevant bases.

Already on `maintenance` branch, soon on `devel`, will be rolled out with the 1.3.5 release.

lf-, 2017-07-24:
The fixed command does indeed work. Thanks for fixing the logging as well.

commit 10e834f6 (2017-07-24): Improve logging of async system commands

Closes #2025
closed 2017-07-24 by lf-