OctoPrint/OctoPrint #2026: M600 times out and fails prints

screwyluie, 2017-07-23:
First and foremost, I'm using octopi. I reported it there and was directed here, which makes sense because this seems like an issue with the core code of octoprint and not something specific to octopi. Sorry if I'm wrong, just let me know.

#### What did you expect to happen?

I expected that octoprint wouldn't timeout while waiting for the m600 command to complete

#### What happened instead?

the m600 command was issued and octoprint triggered a communication timeout while waiting for the command to complete, which failed my print.

#### What were you doing?

I added an m600 color change in my gcode on a long print.
I went to bed knowing that the perpetual beeping of the machine would wake up eventually when it was waiting for a new color. 
I woke up, changed the color and clicked continue on the printer. 
It proceeded to load the new color and go back to where it left off but then nothing... just sat there and oozed on my print. 
Checked back with octopi and is was telling me the connection to the printer had timed out and my print was now failed.

screwyluie, 2017-07-23:
well hopefully someone other than a bot will change the ticket type because nothing else in the template applies imo

oh what the hell I'll paste it all in there anyway and maybe the bot will leave me alone.

JohnOCFII, 2017-07-23:
@screwyluie - Check your "long running commands" list under Settings (Advanced) and make sure M600 is in the list.  You might want to add additional longer-running commands too, to ensure the printer won't time out.
[image]

screwyluie, 2017-07-23:
I did check that and it is in there, looks similar to yours actually

`G4, G28, G29, G30, G32, M400, M226, M600`

ntoff, 2017-07-23:
The information in the logs is vital to understanding what's going on between the printer and the host. Did the printer shut itself down after being idle too long? Did octoprint assume it went offline? No way to know without those logs. Please provide *all* the information asked of you by the issue template.

screwyluie, 2017-07-23:
I can't give you what I don't have but the answers are easy enough. The printer doesn't shutdown nor does anything else. The last thing the log would show is the m600 command being sent. Then like other long commands it waits for the printer to respond and if you wait long enough it just times it and disconnects from the printer. That timeout is the issue. I don't believe it should time out the whole connection while waiting for the printer to respond.

I have provided all the information I have and I believe all the info that is needed.

ntoff, 2017-07-24:
>and I believe all the info that is needed.

If that were the case, it wouldn't be requested in the first place. Unfortunately, that information IS required. Did the printer acknowledge the transmission? Is there a missing `recv: ok` that caused octoprint to stall out? Exactly how long was it from the time the M600 was initiated, until you actually got to it? Does your printer send out any sort of messages during that paused state, messages like "wait" or "busy"?

The logs are always vital and necessary to debug communication and disconnect issues, without them, nothing can be done.

screwyluie, 2017-07-24:
> without them, nothing can be done.

well I'm certainly not going to sit here and argue with you about it. I can't give you what I don't have. No matter how many times you say it, it's not going to happen. This incessant demanding of something that doesn't exist is not helping.

> Did the printer acknowledge the transmission? 

no, as I said the last thing you would've seen in the log, if I had one, would be the M600 command. No other exchanges take place until the command steps are fulfilled. This is the entire problem and what I'm asking to be fixed. Need the connection to not timeout while waiting on the M600 command to finish.

>Is there a missing recv: ok that caused octoprint to stall out? 

yup, that is the problem and what I'm asking to be changed.

>Exactly how long was it from the time the M600 was initiated, until you actually got to it? 

unknown, but longer than the default 30s

>Does your printer send out any sort of messages during that paused state, messages like "wait" or "busy"?

again. The last thing is the M600 command. There is nothing else until the command is done. The last entry in the log would've been M600. There is no OK. There is no Wait. There is no Busy. There is no temperature pings. The last and final line of the log, if I had one, is 'M600'. 

That is the problem. That is the request I'm making. I would like and exception for the M600 command to not timeout the entire connection simply because I took too long to change the filament. It makes no sense to have this feature. What if you're not home, why shouldn't the print wait for you to get there and change the color?

Do you require any further explanation? perhaps I can word it another way if I think about it some more.

foosel, 2017-07-24:
> I can't give you what I don't have.

Except that you have it, and the contribution guidelines also tell you how to get it. But let's do this again here.

Disconnect from your printer. Do this:

[image]

Reconnect to your printer. Reproduce the problem. Just sending an `M600` via the "Terminal" tab might already suffice, if it doesn't you'll need to run a (small) print to reproduce it.

Then do this:

[image]

Attach the downloaded `octoprint.log` and `serial.log` to this ticket. Or upload them somewhere and paste the link. I need this information. No, your description is not sufficient, I need the logs, they are vital to gain insight into what your instance of OctoPrint was doing when the issue arose.

>  I would like and exception for the M600 command to not timeout the entire connection simply because I took too long to change the filament

That exception is already in place and is exactly what that "long running commands" setting does. Apparently it isn't working here for some reason. We need more details to figure out why.

As a side note, I've brought it up repeatedly to the firmware devs that having `M600` not produce any kind of output in stock configuration is a bad idea exactly because it is completely impossible for hosts such as OctoPrint to distinguish between "filament change still going on" and "printer died".

screwyluie, 2017-07-24:
I know where the logs are, but since it happened I have had to reformat and wipe my RPi so I don't have the log. I was not kidding.

> That exception is already in place and is exactly what that "long running commands" setting does. 

I mean, that's what I figured but then this happened and then I thought perhaps even they have a timeout, just much longer.

> As a side note, I've brought it up repeatedly to the firmware devs that having M600 not produce any kind of output in stock configuration is a bad idea exactly because it is completely impossible for hosts such as OctoPrint to distinguish between "filament change still going on" and "printer died".

yeah, the fact that the printer doesn't even send the usual temperature pings is weird to me, but I don't write firmware so maybe there's a reason? I dunno. I completely understand and sympathize, I just don't want to fail another print because I didn't respond to the m600 fast enough.

Assuming I can recreate the same problem I will attempt to get a new log. I'll setup a small print with a color change and see if I can get it to timeout again.

foosel, 2017-07-24:
As said, it might already be enough to just trigger an `M600` via the terminal tab without a print running. That should be easily done.

And no, there isn't a reason that there's no output generated by the firmware there. It's a simple (and sadly all to common) oversight on the firmware dev's part. This keeps happening with commands all over and it's annoying as hell.

Alternatively you might want to also ping your firmware vendor to pull in the `busy:` protocol changes from mainline Marlin (Prusa Research Marlin fork, right?). Feel free to ping me in a ticket on their tracker, I'll happily rant a bit on this topic there. I'm frankly sick of having to somehow work around idiotic firmware behaviour all the time ;)

screwyluie, 2017-07-27:
Ok sorry for the delay, life is busy and all that. Here are the logs from me inducing the problem again. 

Also I noticed in the settings "Max. consecutive timeouts during long running commands" which seems like it would solve the problem but I'm unsure if I should set it to '0' just to avoid this one issue and if that will have any appreciable negative side effects? Thanks.
[serial.txt](https://github.com/foosel/OctoPrint/files/1181335/serial.txt)
[octoprint.txt](https://github.com/foosel/OctoPrint/files/1181336/octoprint.txt)

foosel, 2017-07-28:
Ah! Ok... Now that all makes sense. I had actually managed to forget that I also had implemented maximum consecutive timeouts for long running commands. Yes, settings that to 0 should solve your problem. The negative side effects would be that if your printer crashes during any of your configured long running commands, OctoPrint won't notice and just sit there waiting for ever.

foosel, 2017-07-28:
Now look at that, I actually had opened a ticket for this kind of silent treatment in the Prusa firmware already. See prusa3d/Prusa-Firmware#67. 

Until that's fixed in the firmware, you'll need to disable consecutive timeout detection on long running commands. Please test if that solves your problem and if it does, close the ticket, thanks!

screwyluie, 2017-08-25:
done some testing and as far as I can tell that setting fixes the issue, I haven't been able to repeat it since changing it. thanks.

closed 2017-08-25 by screwyluie

thomasjachmann, 2018-11-10:
For anyone coming here looking for a workaround to this problem: Setting the consecutive timeout detection to 0 didn't work for me. I got a connection error right after the first timeout period. But setting it to a very high value (999) worked.