OctoPrint/OctoPrint #2027: SPI interfacing issue

spreedated, 2017-07-23:
Hi @ll,

i couldn't find any information about the SPI interface and octopi, is it somehow disabled generally?

What I try to do is controlling a WS2801 RGB-LED stripe with the same raspi i'm using to print with.
Actually it's no big deal and i'm runnig these WS2801 on several other raspis without issues.

So, are the GPIO-SPI-pins(10,9,11 or board 19,21,23) disabled for some reason?

I'm using the Adafruit-WS2801 python library, no errors, but no light.

FHeilmann, 2017-07-23:
It's a game of chance to drive WS2801s from a pi directly, since they are 5V but the pins on the pi only output 3.3V. So it may be that you are lucky on your other pis, but unlucky on this one.

Pins are not disabled for sure though, I am running WS2812Bs on my pi without issue.

spreedated, 2017-07-23:
You are misinformed... like i said, im running WS2801 on serveral PI's without any issues, also as an "ambilight" on my TV. All Raspberrys has two +5V pins - the advantage of WS2801 over WS2812Bs is that the PI can compute the information faster, e.g. ambilight.

FHeilmann, 2017-07-24:
> The Raspberry PI GPIOs use 3.3V logic levels, and can be damaged if connected directly to 5V levels (as found in many older digital systems) without level-conversion circuitry.

source: http://elinux.org/RPi_Low-level_peripherals#Introduction

While there are 5V **source pins** on the GPIO header, whatever pin you use to transmit the signal is 3.3V. **Like I said**, in most cases, the tolerance in both the PI's output voltage spec and the WS2801Bs input spec, will allow you to use the signal pin as-is without issue. However, in some cases, the 3.3V logic level of the signal pin can cause problems, resulting in the LED strip not being able to read the logic levels properly.

I am making this point so you have something else to check, in case you checked everything else and couldn't find any issues. Try the LED strip not working on this PI on one of your other PI's and try one of your working strip on the PI you are running octoprint on.

Since you are driving your LEDs via SPI I would check whether SPI is properly enabled in the distribution you are running: [This is a good starting point](http://raspi.tv/how-to-enable-spi-on-the-raspberry-pi)

In any case, this is not controlled by OctoPrint at all, so you either have to take this issue to the OctoPi guys (if you use it) or configure your distro properly. In the latter case it is merely a configuration issue and not a bug.

foosel, 2017-07-24:
As already said, not an OctoPrint issue, closing.

closed 2017-07-24 by foosel

spreedated, 2017-07-24:
This has gone way offtopic. All I wanted to know was "is SPI in general disabled, for some reason, in octopi". - nothing more.

You keep talking about iso layer model, 3.3V pin outputs and so on, which was never the point of discussion, since everything works fine with the same hardware, same configuration, but different software (octopi img), and you want me to check the configuration of SPI in my distro, when im using your distro?!

however, thanks for your 2 cents

FHeilmann, 2017-07-24:
I was merely trying to offer advise as to possible sorces of your LED strips not working.

From your comment I'm inferring you'Re using OctoPi, the distribution that bundles octoprint. Maybe they can help you out.

Kunsi, 2017-07-24:
spreedated, you're not using foosels or "our" distro. Octoprint is a Software running on a computer, like your browser on your own computer. Octoprint does not have control over anything controlled by your OS.

If we'd take this bug report to another system, you'd blame Google Chrome (=Octoprint here) devs because Windows (=OctoPi/Raspbian here) doesn't want to talk to your printer (=LED strips here).