OctoPrint/OctoPrint #2028: [Request] Navbar sticky to the top

EddyMI3d, 2017-07-24:
Hello!
Is it possible to make the navbar sticky to the top? 
Maybe selectable if sticky or not in the settings.

Thank you,
Eddy

token47, 2018-01-07:
+1 on this. It's _really_ useful to have it always on top, specially when you have all kinds of info on it (by means of plugins that use it to show more info).

This probably should be a plugin itself, and it probably is very simple to make (add some HTML/CSS commands in the right place), but unfortunately my coding skills were not enough to do it.

jneilliii, 2018-01-07:
This can be accomplished with 2 lines of javascript code.

```
$("#navbar").toggleClass("navbar-fixed-top navbar-static-top");
$("div.container.octoprint-container").css("margin-top","60px");
```

jneilliii, 2018-01-07:
I threw together a quick plugin for this.  You can find it at the link below.

https://github.com/jneilliii/OctoPrint-FloatingNavbar

token47, 2018-01-07:
That is very nice. Thank you.

I guess you know, but anyway: it does not appear on the search inside octoprint nor on the search on the site, yet.

Will try, thanks!!!

jneilliii, 2018-01-07:
I have submitted my plugin to the repository rust show up in plugin manager but it hasn't been accepted yet.

kantlivelong, 2018-02-20:
A plugin is now available to do this. See https://plugins.octoprint.org/plugins/floatingnavbar/

Closing

closed 2018-02-20 by kantlivelong