OctoPrint/OctoPrint #2029: Current tool height

MrBalonio, 2017-07-25:
Hello, 

I been looking at the API documentation but I cannot find how to head the current tool height.

Is there a way to get this information. 

I am building a python script that adjusts the ptz of my camera based on printer height.

Thanks.

jorender, 2017-07-27:
@MrBalonio this took me a little bit to figure out as well.  I'm not sure if it's the only or even the correct method but it works for me.  You'll want to register a callback in your view model for onEventPositionUpdated.

Callback documentation [here](http://docs.octoprint.org/en/master/plugins/viewmodels.html?highlight=onevent) lists the onEvent&lt;EventName&gt;(payload) signature.  The link to the list of available events will list items you can replace &lt;EventName&gt; with.

foosel, 2017-07-27:
For questions or other kind of help or support, please get in touch on the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2017-07-27 by foosel