OctoPrint/OctoPrint #2030: [Feature request] filename customization for timelapse videos

gege2b, 2017-07-25:
Hi there

I came accros a (little but annoying) not-a-bug thing today about timelapse.
Current filename pattern for rendered timelapse is "GCode filename_timestamp.mpg"

It's pretty good as is, but if (like me) your gcode's filename begin with a timestamp, you end up with a filename in the form "20160626-094403-bla-bla-original-name_20170725131201.mpg"

If you slice & print in a row, it's ugly, but not so annoying.

But if you print a file, sliced and sent a few (months|years) ago, it will not be shown where expected in the list. In fact, you'll need 

Adding a simple text box in the timelapse settings would make the job (may be along with some usefull placeholder like {timestamp}, {time_start}, {time_end}, {file_size}, {duration}... or whatever)

Sunoo, 2017-07-26:
Possibly related to this, but it would be nice to have a way to only keep timelapses for successful prints, or at least mark the failed prints in some way.

gege2b, 2017-07-26:
Well, no really related I think

Personnally I prefer to keep the timelapse of the failed print, so I could see when and how things gone wrong :)

Sunoo, 2017-07-26:
It would be nice to at least see -failed at the end of a file name though. Trying to sift through timelapse files becomes a mess quickly.

gege2b, 2017-07-26:
there is a brainstorm ticket about timelapse ( #1980 ) witch could be interesting for that kind of features

gege2b, 2018-05-15:
I think I can close it too, Octolapse support both of this features now 
@foosel feel free to reopen it if needed

closed 2018-05-15 by gege2b