OctoPrint/OctoPrint #2032: Octoprint does not connect to GRBL laser engraver

scottmil, 2017-07-26:
#### What were you doing?

I was trying to connect Octoprint to a GRBL Arduino clone-based laser engraver that utilizes a CH340/341 USB communications chip.  The Raspberry Pi has a CH341 kernel driver. 

Ideally provide exact steps to follow in order to reproduce your problem:

1. Connect USB cable between Raspberry Pi running Octoprint and laser engraver.
2. Set serial port to /dev/ttyUSBO, baud to 115200, then press Connect.
3. Laser engraver initially connects and responds, then Octoprint sends commands the engraver interprets as an error.

#### What did you expect to happen?
Octoprint would successfully connect to laser engraver.

#### What happened instead?
Laser engraver eventually responds with its error code 9 and causes a disconnection. See attached screen shot.

#### Did the same happen when running OctoPrint in safe mode?
No. Instead of causing the engraver to generate an error, Octoprint timed out the connection to the engraver.  See attached screen shot.

#### Branch & Commit or Version of OctoPrint
Octoprint 1.3.4 (master branch)

#### Operating System running OctoPrint

Raspberry Pi Raspian Jessie version 4.9.35-v7+
OctoPi's version 0.14.0

#### Printer model & used firmware incl. version

Homemade Arduino clone-based laser engraver running GRBL 1.1f

#### Browser and Version of Browser, Operating System running Browser

Safari 10.1.1, Mac Sierra 10.12.5

#### Link to octoprint.log

https://gist.github.com/scottmil/cdd2fc37524e6ba37f071188cfd8455d

#### Link to contents of terminal tab or serial.log

https://gist.github.com/scottmil/a5e91aa02a92c293bbc18115ea8280fd

https://gist.github.com/scottmil/794c0e913666b8f2eb4f429cb02ef3cd

#### Link to contents of Javascript console in the browser

Not applicable, not a GUI issue.

#### Screenshot(s)/video(s) showing the problem:

Terminal Screenshot
Terminal Screenshot - Safe mode
lsmod and lsusb responses

I have read the FAQ.
[image]
[image]
[image]

ntoff, 2017-07-26:
error9 is because grbl is locked and in an alarm state (probably due to it not being homed yet, being in an alarm state likely has nothing to do with octoprint) and octoprint is sending it gcode.

Tell octoprint not to disconnect on errors, and ignore unhandled errors

[image]

Untick SD support (stops M21) and turn off feature auto detection (stops it sending M115)

[image]

foosel, 2017-07-26:
Also keep in mind that OctoPrint is tailored towards 3D printers, whose firmware while looking pretty identical to CNC GCODE in fact is NOT exactly that and stuff will not behave as expected out of the box so even with the above it's unlikely you'll be able to use OctoPrint out of the box with your laser. E.g. OctoPrint constantly polls the printer's temperatures when connected through `M105` and that will most likely produce an error here too.

Take a look at the [GRBL plugin](http://plugins.octoprint.org/plugins/octoprint-grbl-plugin/), that might solve this for you.

Closing because what you are seeing is more or less expected behaviour.

closed 2017-07-26 by foosel

foosel, 2017-07-26:
@ntoff just gave me the friendly hint that you already are using the Grbl plugin 🤦😳 Sorry, I need a vacation. 

In any case, I suggest following his advice (and maybe sending a PR to the Grbl plugin to update its docs to also include the above configuration changes).

ntoff, 2017-07-26:
Oh and an addition to my above tips, in case you haven't already (don't recall seeing M5 in your serial log) change your start command to M5 and disable checksums as per the instructions on the grbl plugin git page. https://github.com/mic159/octoprint-grbl-plugin#required-configuration