OctoPrint/OctoPrint #2033: [RC Feedback] Feedback on the 1.3.5rc1 Release Candidate

foosel, 2017-07-26:
Please provide *general* feedback on your experience with the 1.3.5rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

As a quick additional heads-up: Starting this friday I'll be on vacation for the next three weeks. I'll tackle all feedback and questions once I get back and then decide whether I need to push out a second release candidate or can directly jump to 1.3.5 stable. But don't expect any replies from me in the meantime, I need some actual time off for a change ;)

ctgreybeard, 2017-07-26:
Using Google Chrome Canary I don't see any data mousing over the temperature graph. I do see the vertical red line but no numbers.

foosel, 2017-07-26:
@ctgreybeard Hah... see? That's why we have RCs ;) Something I did since implementing that feature must have broken it a bit again. It re-uses the legend to display the values for the current location of the red line, but that doesn't seem to work right now. 

Ah well, I figure I'd probably need a second RC anyhow down the road, especially considering that this time you all will have a some more time than usual to try to break things ;)

goeland86, 2017-07-26:
Not tried a print yet, but otherwise things look ok. Going to remove the dark theme and see if I notice anything broken. (BTW the data mousing temp works on Firefox...)

chippypilot, 2017-07-26:
Thanks for the update.
I am testing on a Pi Zero W and windows 10 via Chrome, Edge and Firefox browsers. I've connected to the printer, but not printed yet. In general the various pages are working. But I can confirm that the temperature values are not changing when moving the red line over the chart in any browser.

goeland86, 2017-07-27:
Just a heads up - I've now had to restart my replicape (where octoprint is installed) and, for some reason, ran into an error on boot about "port already in use"... Only octoprint uses port 80, and the error disappeared after reverting to 1.3.4. I know this is a quick heads up and I don't have full logs, but I needed to get a print done for a customer last night.

Will try to get a proper bug report to you ASAP. It could be due to the fact that on Kamikaze/Umikaze we're installing Octoprint "natively", i.e. not in a python sandbox, but like I said - will try to get a proper log for you when I get my print part orders out the door.

roygilby, 2017-07-27:
Just started testing 1.3.5 RC1 on a Pi 3B.  Under the "Rendering Options", "Zoom in on Model" does not do what we expect when that option is selected.  It worked OK on previous version of Octoprint.  Using Firefox Version 54.0.  I don't have any alternative browsers available today, so can't test if this problem is Firefox only.

Next a question ......

I'm a retired Test Analyst, and have spent a few years doing Testing of Web Based Apps before retiring.  Do you have a specific set of test cases that you'd like tested on this release ?   

Roy Gilby
Brisbane, Australia.

foosel, 2017-07-27:
@goeland86 I appear to have introduced that issue by moving the gcode interpreter into its own subprocess. For some reason it apparently keeps the port of the intermediary server ("The server is still starting up...") open if it has to analyse a file during startup (from the backlog), which in turn causes issues when proper Tornado tries to start. This is tracker in #2035. I'll probably try to get 1.3.5rc2 out for this reason today.

@roygilby 

>Under the "Rendering Options", "Zoom in on Model" does not do what we expect when that option is selected.

What does it do instead? :) A screenshot would be awesome. 

No specific test cases in mind btw, rather a general "use it like you'd normally do, give feedback on stuff that looks odd or broken" as well as "and if everything looks fine, please say so too" :)

TylonHH, 2017-07-27:
While Updating I got [some Warnings](https://pastebin.com/y8hnpD2J).

Kunsi, 2017-07-27:
On my laptop, the new temperature selector is wrapped onto two lines.

[image]

foosel, 2017-07-27:
@TylonHH that one I already found and fixed yesterday. Nothing critical, just stupidity ;)

@Kunsi Gnaa... What browser is that?

TylonHH, 2017-07-27:
Same here with Browser and two lines: Firefox 54.0.1 (64-Bit)

Kunsi, 2017-07-27:
Ah, missed that. Firefox 54.0.1 (64-bit) too, on Mac OS.

TylonHH, 2017-07-27:
One thing I had in previous version also: I have an dual extruder with one nozzle. I told it Octprint.
But on the temperatur tab the second sensor is useless because there is no second sensor when sharing one nozzle.
[image]

Or shall I open a request (issue)?

foosel, 2017-07-27:
@TylonHH Bugs in general should always get their own bug reports, so please file a full report for that. Please also include your printer profile. Thanks.

But this will definitely have to wait until 1.3.6: The only things I'll tackle in follow-up RCs for 1.3.5 are any observed regressions or things that are broken in stuff added in 1.3.5. Learned my lessons there ;)

foosel, 2017-07-27:
@TylonHH never mind, stumbled over it by pure chance and decided to fix it. I just hope that won't bite me in the behind later. Fixed by the above commit b82d955

Preparing a new RC right now, will close this ticket with link to the new one once that's done. Thanks to everyone who reported back so far, that's helped a lot!

foosel, 2017-07-27:
Marking as closed because 1.3.5rc2 has been [released](http://octoprint.org/blog/2017/07/27/new-release-candidate-1.3.5rc2/). Comments are still possible though.

New feedback ticket is #2038

closed 2017-07-27 by foosel