OctoPrint/OctoPrint #2036: FAILURE TO AUTODETECT USB CONNECTION

NervenCid, 2017-07-26:
Hello, i'm new into using octoprint, i beg your pardon but i would like to report a malfunction in the serial port comunication detection (and it had been working fine the last week):

I upgraded to the latest version of octorpint, and when i tried to connect to serial port with automatic settings i got:

Unhandled communication error
The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: Failed to autodetect serial port, please set it manually.

The terminal log that i get is this:

Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.

When i use manual setting for ' /dev/ttyACM0' i got:

Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0xa2d9b610, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=120.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: echo:Marlin 1.1.Updated: 2016- Author: (none,t config)
Recv: Frardcoded Default Settings Loaded
Recv: echo:Steps per unit:
Recv: Z3200.00 E250.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M201 X3000 Y3000 Z100 E10000
Recv: echo:AcceR=retract and T=
Recv: ec0.00
Recv: echo:Advanced variat time (ms), X=maximum XY jer0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset (mm)
Recv: 4.00
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M2
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor

When i connect directly to a laptop the printer is working fine.. ¿what is the problem?

Thank you

lf-, 2017-07-27:
What if you do something like `picocom -b 115200 /dev/ttyACM0` and enter exactly what OctoPrint tried to send: `N0 M110 N0*125` and see if there is a line that says `ok`. OctoPrint, in this error, is reporting that it didn't see one.

If it takes a while before it shows up, consider [changing your serial timeouts in the settings](http://docs.octoprint.org/en/master/configuration/config_yaml.html#serial). It seems that the relevant one is `connection`.

Kunsi, 2017-07-30:
Please provide all the requested information. For connection related issues, ALWAYS provide a serial.log. If it works using some other Computer, please also provide a log and connection settings of that software.

Using the provided information, I can see OctoPrint is waiting for your printer to reply 'ok', which it doesn't send (or too late).

NervenCid, 2017-07-31:
@Kunsi Thank you for your anwer... i beg your pardon

But where can i find serial.log? I'm new on linux OS

Thank you

lf-, 2017-07-31:
@NervenCid In the big yellow box at the top when you create an issue, you are told to read [this document](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). It contains the location of the log file.

NervenCid, 2017-08-02:
Following the template provided:

#### What were you doing?

1. I turn on the printer and the raspberry pi
2. I waited till the server is online
3. I logged to Octoprint
4. I pushed the button "connect" with AUTO settings
5. I got a connection error and a red pop up appears with this message

 Unhandled communication error
The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: Failed to autodetect serial port, please set it manually.

#### What did you expect to happen?

Connect to the printer

#### What happened instead?

It appearde the pop-up with this message:

Unhandled communication error
The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: Failed to autodetect serial port, please set it manually.

#### Did the same happen when running OctoPrint in safe mode?

I tried different configurations, but the error is aleatory

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint

Client: Windows 10

Server:

[OctoPi, Linux, Windows, MacOS, something else? With version please,
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version

Is a custom XY Core printer

Firmware: Marlin 1.1

Hardware: Arduino+RAMPS

#### Browser and Version of Browser, Operating System running Browser

Google Chrome latest

#### Link to octoprint.log

https://gist.github.com/NervenCid/f4cf79336a9bbf31279fe50bf988f8ff

#### Link to contents of terminal tab or serial.log

serial.log is empty

#### Link to contents of Javascript console in the browser

https://gist.github.com/NervenCid/156e537f0fd3873f3b700ab88fa62371

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

[logs.zip](https://github.com/foosel/OctoPrint/files/1194846/logs.zip)

Kunsi, 2017-08-02:
You need to specifically enable serial.log before it gets populated. serial.log is the relevant part here, so we absolutely need it.

markwal, 2017-08-03:
From the short snippet you have from the terminal window it seems clear that you are dropping bytes from the printer to OctoPrint.  You may want to try 115200 baud rate for grins, but I suspect you've got either a noise problem or a power problem.

Noise can be exacerbated by a long or thin usb cable running parallel to the stepper motor cables (they're noisy).  You can try a shorter thicker USB cable with a ferrite choke on it.

Power can be a problem because if your power sags you can mess with the communication chips that are trying to talk over the USB wire.  This can easily be intermittent with a long thin power cable since its resistance will rise whenever the Pi draws more current (like when talking over wi-fi).  Key here is to have a solid power supply of sufficient amperage rating and a nice short, thick cable to the pi.

NervenCid, 2017-08-17:
@Kunsi @markwal sorry for my delay (i don't have the printer ner of me) the missing file serial.log is here:

https://gist.github.com/NervenCid/bc7431c16d125a20c69df3b09f8f1dd7.js

I changed the baudrate to 115200 but the problem persists... i tested the arduino with a laptop offline but is working fine

Thank you

markwal, 2017-08-17:
That is octoprint.log. Serial.log is different and is off by default since it's io intensity can exacerbate the symptoms you're trying to solve, but would still be very helpful to understand.

foosel, 2017-08-21:
Nah, it actually is `serial.log`.

```
Unexpected error while connecting to serial port: /dev/ttyACM0 SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_detectPort:1705
```

This appears to be the problem here. It can't connect because either the printer suddenly vanished while attempting to connect, there was concurrent access to the same serial port from another program or - what @markwal already pointed out - there's some serious communication/brown-out issue going on here.

Do all your connection attempts look like this or are the other variants as well?

While things don't even get as far here that the baudrate would matter, in general you'll need to choose the baudrate that matches your printer's firmware. Usually it's either 115200 or 250000 but some printers have other values here. Still though, you can't even open the serial port successfully here, so that's not the problem yet.

foosel, 2017-08-21:
Never mind, I just found the snippets further up with some more examples (they were a bit hidden due to not being encased in triple `, which formats output like that in a way it's way better readable - I took the liberty to edit the comments to fix that).

So first of all, it looks like 250000 baud is the correct speed for your printer, otherwise you wouldn't have gotten readable output in one of your samples at all. That output however hints at some corruption:

```
Recv: echo:Marlin 1.1.Updated: 2016- Author: (none,t config)
Recv: Frardcoded Default Settings Loaded
Recv: echo:Steps per unit:
```

See how OctoPrint received "Frardcoded" - that was probably rather "Fr-something" and "Hardcoded" and got mangled in transport. As do a lot of the other lines:

```
Recv: echo:AcceR=retract and T=
Recv: ec0.00
Recv: echo:Advanced variat time (ms), X=maximum XY jer0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset (mm)
Recv: 4.00
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M2
```

For reference, it's supposed to look more like this:

```
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo:Marlin bugfix-1.1.x
Recv: echo: Last Updated: 2017-05-04 12:00 | Author: @foosel
Recv: Compiled: Jul 12 2017
Recv: echo: Free Memory: 2019  PlannerBufferBytes: 2528
Recv: echo:V38 stored settings retrieved (453 bytes; crc 32061)
Recv: echo:  G21 ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X160.00 Y160.00 Z8000.00 E210.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z2.00 E200.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E3000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P800.00 R1000.00 T1000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E1.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H210 B70 F0
Recv:   M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P19.18 I1.36 D67.42
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.21
Recv: echo:SD card ok
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: echo:SD card ok
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:BQ Hephestos 2 EXTRUDER_COUNT:1 UUID:8d083632-40c5-4649-85b8-43d9ae6c5d55
Recv: Cap:EEPROM:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: ok
```

So that points at underlying communication issues which might explain what you are seeing - again with multiple possible causes, see above and also [this very important FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception).

I also spotted this in your `octoprint.log`:

```
2017-08-01 12:43:09,884 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Heating failed, system stopped! Heater_ID: bed

| Last lines in terminal:
| Send: M105
| Recv: ok T:20.0 /0.0 B:92.2 /100.0 @:0 B@:127
| Send: M105
| Recv: ok T:20.0 /0.0 B:92.2 /100.0 @:0 B@:127
| Send: M105
| Recv: ok T:20.0 /0.0 B:92.3 /100.0 @:0 B@:127
| Send: G91
| Recv: ok
| Send: G28 Z0
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: Error:Heating failed, system stopped! Heater_ID: bed
```

Was that from a different printer or did you get a connection going after all at one point?

NervenCid, 2017-08-23:
Thank you for your answer mrs @foosel i tried sugestions on the FAQ you provided:

1) I am providing power to the raspberry pi with a power source of 12V 90 A directly (with a conversor 12 to 5V) and i can connect to the pi via ssh besides the serial connection failed

2) I changed to a cable with double filterand supposely "shielded"  but is so long i will try it anyway

3) i don't know if i can upload photos to github

4) How could i know if a additional program is trying to access to the printer?

I could'nt tested yet because we are having problems with the network in my university

foosel, 2017-08-25:
> 1. I am providing power to the raspberry pi with a power source of 12V 90 A directly (with a conversor 12 to 5V) and i can connect to the pi via ssh besides the serial connection failed

Does your *converter* also provide a *stable* 5V/2-3A? Because the best 12V supply behind it doesn't help you if it isn't actually providing a solid converted 5V. Fluctuating voltage and/or current on the Pi power input appear to be the most common reason for these kinds of issues, so best measure and keep an eye on your Pi's red LED.

> 2. I changed to a cable with double filterand supposely "shielded" but is so long i will try it anyway

The shorter the better.

> 3. i don't know if i can upload photos to github

You can. See here:

[image]

> 4. How could i know if a additional program is trying to access to the printer?

That's tricky. If you are using OctoPi it shouldn't. But as said above, the most likely cause is insufficient power.

foosel, 2017-10-26:
Closing due to lack of response.

closed 2017-10-26 by foosel

NervenCid, 2017-10-27:
Hello Miss @foosel 

Sorry for the delay, but i recently updated to the version 1.3.5

Its fails whe i use autodetect, but its connect when i set up the parameters manually

foosel, 2017-10-27:
Yes, but provide all requested information, otherwise it is impossible to get to the bottom of this and the ticket will be closed again due to lack of information needed.