OctoPrint/OctoPrint #2037: Can't edit or add new printer settings

djarami, 2017-07-27:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
aka the Contribution Guidelines that are linked ^-- just up there
in the big yellow box.

Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ...", "I have a
question ..."), that belongs on the mailing list or the G+ community
(both linked in the "guidelines for contributing" linked above, read
them!), NOT here.

Mark requests with a "[Request]" prefix in the title please. For bug
reports fully fill out the bug reporting template (if you don't know
where to find some information - it's all described in the Contribution
Guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete ANY lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to report is still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the Contribution
Guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?
I updated some settings in the printer config and I changed the origin to 0 instead of center.  That caused some other error later when I went to print saying that things were out of bounds.  So I tried to go back and change it, the edit pencil is greyed out.  

I tried adding a new printer, but the confirm button keeps circling even when I haven't clicked on it.

I tried restarting octoprint, nothing.
I tried rebooting and still not able to edit the printer config.

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. ...  Settings icon, Printer Config
2. ...
3. ...

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.5rc1 (rc/maintenance branch)

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version
ANet 8 - Skynet3D
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome, Mac OS Sierra 10.12.5
[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-07-27:
Could you please provide _all_ requested information? Your ticket is missing a lot of what is needed to further look into this. Thanks.

djarami, 2017-07-27:
That's a lot of information.  I tried to get as much as I could. I watched your video and got excited about your new RC.  I'm new to octoprint and love it. I'm not new to development so I understand why you want all that.  I just wanted to let you know something weird is going with the printer settings. 

One thing I can't seem to find is how to back out the RC and go back to the general release.  I'm very excited about the new function.  

I also noticed that when I out it safe mode, things work fine.   I don't know if that means one of the plugins are affecting things or if there is something else in the system. 

Btw - enjoy your upcoming vacation. 

I look forward to learning more about your system and being able to collaborate more.

djarami, 2017-07-28:
I just updated to RC2, same problem.  I was able to edit once and doesn't let you edit again even when you restart octoprint.  It works fine under safe mode.  I'll take a look at the JS Console log in a little bit - that could have something.

foosel, 2017-07-28:
If it works fine under safe mode, it's one of your third party plugins that's causing the issues. Safe mode disables all plugins that do not come bundled with plugin, so it basically turns your OctoPrint instance into kind of a stock install again while its active.

What plugins do you have installed?

And yes, the JS console might give a hint here. What you are describing sounds like something gets confused in request processing.

foosel, 2017-10-02:
Anything new here?

closed 2017-10-11 by foosel