OctoPrint/OctoPrint #2038: [RC Feedback] Feedback on the 1.3.5rc2 Release Candidate

foosel, 2017-07-27:
Please provide *general* feedback on your experience with the 1.3.5rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

As a quick additional heads-up: Starting this friday I'll be on vacation for the next three weeks. I'll tackle all feedback and questions once I get back and then decide whether I need to push out a second release candidate or can directly jump to 1.3.5 stable. But don't expect any replies from me in the meantime, I need some actual time off for a change ;)

JohnOCFII, 2017-07-27:
TL;DR - 1.3.5rc2 is working well for me.

Used Safari 10.1.2 on MacOS 10.12.6 for the following:
- Upgraded from 1.3.4 to 1.3.5rc1 yesterday.  This morning upgraded to 1.3.5rc2 on a Raspberry Pi3
- Uploaded GCODE file sliced with S3D 4.0 from my Mac to internal OctoPrint SD card
- Connected to Prusa i3 MK2 running firmware 3.0.12-rc2
- Re-set from Z-change to timed timelapse (as I was printing in vase mode)
- Tested Z-axis control from Control tab
- Used pre-sets to pre-heat bed and hot-end
- Selected print job and initiated print
- Monitored temperature, including seeing the numbers reflect the red-line location accurately
- Regularly switched between Temperature, G-Code Viewer, Control and Terminal tabs during 2 hour print.
- After print completion, successfully uploaded, then deleted timelapse movie
- Print History Plug-in and OctoSlack Plug-ins both worked as expected
- Successfully disconnected OctoPrint from printer

- I also connected and viewed Control and Temperature and G-Code tabs from Mobile Safari on both iPhone 6S and iPad Pro 10.5 running iOS 10.3.3.

John

chippypilot, 2017-07-27:
All is working fine for me. uploaded a file and printed. 
Using RPi Zero W. Browsers: Chrome, FF and Edge on W10

thanks

roygilby, 2017-07-27:
All working OK for me.  Tested on:

- RaspberryPi 3b running OctoPi
- Browsers Tested;  Firefox version 54.0 on Linux Mint, and Chrome Browser Ver 60.0.3112.78
- Printer:  Tevo Tarantula running Marlin Ver 1.1.4 customised

Test Cases:

1. Load Gcode File
2. Preview in Gcode Viewer.  Test Centre Viewpoint and Zoom. Test general functionality.
3. Test Temperature Settings.
4. Test "Controls" Tab.
5. Check Terminal Tab. Issue a few test commands.
6. Print Test Gcode File.
7. All successful.

Seems good so far, will conduct further testing during general use today.

Roy Gilby

Kunsi, 2017-07-28:
Just to note, https://github.com/foosel/OctoPrint/issues/2033#issuecomment-318279381 (temperature controls in two lines) is fixed. Regular use also works fine for two small prints, haven't tested any larger prints yet.

foosel, 2017-07-28:
Thanks all of your for your extensive feedback so far! :heart:

Looks like I can go on vacation now without having to worry too much ;) See you on the other side :)

cxt666, 2017-07-29:
Testing succesfully on RPi2 running Octopi with Anet A9 & Telegram Plugin. I did 3 prints with: 

- Uploading & Deleting .gcode
- setting Temp Offset (That's working very nice now, bytheway!)
- Moving the Printhead around, extruding, retracting
- Printing uploaded .gcode, pause & abort
- Watching Print in GCode Viewer

fhbmax, 2017-08-01:
The delayed cancellation bug seems to be fixed for me.

goeland86, 2017-08-02:
This pre-release works fine for me. Cheers!

DaSTIG, 2017-08-11:
TL-DR  is working well for me!  Thank you very very much for such a great product and quality work. Its much appreciated. 

not able to replicate, but i did had instance that crashed the server when cancelling a big file (17mb) and i had to manually reset pi and printer. 

if i may provide some constructive feedback...  this comes from usability standpoint when changing nozzles often for multiple materials on the same machine....
being able to mirrored, cloned, moved (whatever) the temparature buttons to the control tab as well?  Maybe a way to either customize your tabs,  or merge them if we wanted to?

I find myself having to keep moving back and fwd between the temp and control tab often... for example....when preheating the machine for a print, or preheating machine after a nozzle change to tighten and calibrate Z height... im always changing nozzles  for different orders if i need to print with a Stainless or hard steel or bigger size....  having those settings there will be great... i understand the use of the temp tab and no need to change anything there... i just feel i have enough space on my browser to have more options there.

Thank you  very much and keep up the amazing work.

andrivet, 2017-08-13:
RC2 is working well. I have done successfully several prints. My setup:

* Wanhao Duplicator i3 plus
* Manual installation of OctoPrint on a Raspberry Pi 3B (Raspbian)
* Slicing with Cura 2.7 Beta. GCODE sent to OctoPrint directly from Cura.
* Browsers: Safari 10.1.2 and Chrome 60 on macOS 10.12.6

foosel, 2017-08-23:
Hello everyone, thanks for your feedback during my vacation! :) Sounds like 1.3.5rc2 is working well for everyone who reported back so far, nice!

I noticed something that looked a lot like #2035 right on Monday when I tried to start my development server up for the first time again - haven't been able to reproduce since though and am starting to think I was imagining things O_O Did anyone run into any more "port already in use" issues with this RC by any chance?

goeland86, 2017-08-23:
I haven't seen it since 1.3.5rc2 was installed, no. Perhaps you had some
cached python files left over that were subsequently over-written when you
forced a restart of the python interpreter? Grasping at straws here, but
since it's been mostly positive reports on that particular issue...

foosel, 2017-08-23:
Just rediscovered the corresponding log files. Looks like a race condition caused by the online check confusing some things and then causing some stuff to run which isn't supposed to run yet... huh... I fear that's something I need to fix. Shoot.

foosel, 2017-08-25:
Marking as closed because 1.3.5rc2 has been [released](http://octoprint.org/blog/2017/08/25/new-release-candidate-1.3.5rc3/). Comments are still possible though.

New feedback ticket is #2095

closed 2017-08-25 by foosel