OctoPrint/OctoPrint #2039: Cannot use second extruder on dual-head printer

morganh83, 2017-07-27:
Support for better control with dual-extruder printers would be excellent. 
I am using a Prusa i3 with Marlin firmware.
What I'm trying to do is use only one of the heads at a time due to material/color preferences for that print. I have tried the following:

Loaded two separate profiles from Cura (tested and working when directly connected to the printer)
Each profile selects a different print head (sets the proper temperatures and uses T0/T1 in the gcode script. 
Below is the gcode script for T0

```
G21     ;metric values
G90     ;absolute positioning
M82    ;set extruder to absolute mode
T0       ;Set first extruder
M106 S255           ;Turn on FAN to prevent clogging
G109 T0 S140       ;Preheat the nozzle
M140 S{print_bed_temperature} ;Set Print bed temp
G28      ;move X/Y/Z to min endstops
G29      ;Auto-Leveling GO!
M109 T0 S{print_temperature}   ;Fully heat the nozzle
G92 E0      ;zero the extruded length
G1             ;Start printing
```

Which gets added to the .gco file after slicing.
The following is the T1 gcode script.

```
G21    ;metric values
G90    ;absolute positioning
M82   ;set extruder to absolute mode
T1      ;use 2nd extruder
M106 S255            ;Turn on FAN to prevent clogging
M190 T1 S140        ;Preheat nozzle
M140 S{print_bed_temperature}    ;Start heating print bed
G28                        ;move X/Y/Z to min endstops
G29                         ;Auto-Leveling GO!
M109 T1 S{print_temperature2} ;Fully heat the nozzle
G92 E0                   ;zero the extruded length
G1                          ;Start Printing
```
This code never gets added to the .gco file. I tried different methods to troubleshoot, including removing all of the other gcode.x scripts from the Cura ini file. It was attempted as start.gcode and start2.gcode. Also, the code with "M109 T1 S{print_temperature2}" I have tried removing the 2 and setting the print temp on the UI as well as setting it to 0 in various combinations with no luck. The following snippet is the .gco code after slicing when trying to use the T1 profile (and gcode script).

```
;Generated from Connector-15mm-20170101-v1.stl 477d24d05ccdf815d69c1e9b3503451738d297bf
;Generated with Cura_SteamEngine 15.04.6
M140 S60.0
M190 S60.0
;Sliced at: Wed 26-07-2017 22:17:22
;Basic settings: Layer height: 0.1 Walls: 0.8 Fill: 20
;M190 S60 ;Uncomment to add your own bed temperature line
;M104 S?print_temperature? ;Uncomment to add your own temperature line
;M109 T1 S205 ;Uncomment to add your own temperature line
;M109 T0 S?print_temperature? ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M107       ;start with the fan off

G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops

G1 Z15.0 F9000 ;move the platform down 15mm

T1                      ;Switch to the 2nd extruder
G92 E0                  ;zero the extruded length
G1 F200 E10             ;extrude 10mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F200 E-16

T0                      ;Switch to the first extruder
G92 E0                  ;zero the extruded length
G1 F200 E10             ;extrude 10mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F9000
;Put printing message on LCD screen
M117 Printing...
```

When printing with T0, the gcode script I added is used and the printer functions normally.
Then attempting to use T1, my gcode is ignored and default code is used, which attempts to use T0 instead, as well as not homing the axis and leveling the bed.

I would love to see another temperature control section for the T1 print temperature and/or a tool selector to print with (T0/T1/Both).

foosel, 2017-07-28:
OctoPrint's built-in slicing is very rudimentary and right now only supports translating a model into GCODE with whatever slicing profile you have selected. You say you 

> would love to see another temperature control section for the T1 print temperature and/or a tool selector to print with (T0/T1/Both).

There's no temperature control section for any kind of print temperature in stock OctoPrint. Are you sure you aren't using the [full features Slicer plugin](http://plugins.octoprint.org/plugins/slicer/)? If so, your request should be directed there.

morganh83, 2017-07-28:
I am using that plugin. I'll direct this there. Thank you.

closed 2017-07-28 by morganh83