OctoPrint/OctoPrint #2040: How to set up multiple WiFi networks to work with Octopi?

milkman333, 2017-07-27:
I have been travelling back and forth with my printer a lot lately and every time I unpack and setup the printer, I have to remove the SD card from the octopi and change the SSID login info, save, re-insert and power on. Not terrible but also less than ideal.

Is there a way to store two different sets of WiFi network credentials so that I don't have to keep editing octoprint-network.txt every time I want to connect to the other network?

Hope that makes sense, let me know what I can do to help

milkman333, 2017-07-27:
I am not sure I understand, I didn't see any other discussion forum for the software

Sorry I couldn't provide more info, hopefully this is enough to get an answer from someone more knowledgeable

ntoff, 2017-07-27:
This is a bug tracker, not a help request hotline. Also Octoprint has no part in setting up your wifi connection, that is handled by the underlying raspbian operating system. This is not the place to ask for help, or request that as a feature. Please see the G+ group or the mailing list linked to you by the bot.

foosel, 2017-07-28:
As already pointed out, this is the wrong location for questions, sorry. 

For questions or other kind of help or support, please get in touch on the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2017-07-28 by foosel