OctoPrint/OctoPrint #2042: [Feature request] Option to change and automatically dismiss pnotify popups

aviatorken89, 2017-07-28:
I searched around and couldn't find any docs about it, but it would be nice if the pnotify popups automatically hid after a user-define threshold time that could be set to whatever you want.

When I log in the pnotify popup does hide after a few seconds, but if I go to OctoPrint Settings > Software Update and click `Check for update now` it just pops up and says: `Everything is up-to-date` and stays there and doesn't go away.

Some kind of global option would be great that you hide each pnotify popup after user defined X amount of seconds after they appear in chronological order.

Currently using the latest devel commits.

foosel, 2017-10-12:
It is actually intentional that some of those popups are persistent until someone closes them while others autoclose (and I think the timeout was more like 5s but I might be mistaken). I see that with the "Everything is up to date" one that's kinda senseless though.

I can change that, but I won't make stuff like update notifications or announcements switch to auto-hide - the whole point of them is to demand attention and action from the user, auto hide might cause them to not be seen if the user opens the page, does something else then returns to the browser only after a minute or so.

klcjr89, 2017-10-12:
So give power users like me the option to change it in the Settings.

foosel, 2017-10-13:
There is no central place to change "it" so that would make every single notification a mess to maintain. I also can't police plugin authors in that regard either, so it would not really fix anything apart from what is in the core application (and as I already said, I'll solve the "Everything is up to date" thing anyhow which appears to have been the trigger of this request). 

If you see a way to have PNotify have a general override of this setting, it would be an option to write a plugin that adjusts the used PNotify library accordingly - something easily be done by everyone open to a bit of coding, not just me, and encapsulated. But I can't add settings for every single bit of behaviour all across the application - that's what produces feature creep, makes stuff pretty much unmaintainable (and maintenance is ALREADY what I spend most  time on apart from issue triage) because everything just drowns in "if"/"else", produces confusing settings that nobody understands anymore and basically boils down to producing a lot of overhead and misery for a way too small advantage.