OctoPrint/OctoPrint #2043: During long prints, it tmp folder runs out of space, it would leave the files there forever

guysoft, 2017-07-28:
#### What were you doing?

I have an OctoPrint 1.3.4 and OctoPi 0.14 installation with timelapse on by default
steps to follow in order to reproduce your problem:

1. Set timelapse on by default
2. Print something that takes a long time which would make your card run out of space

#### What did you expect to happen?

The timelapse should fail, but then clear the tmp folder when you start a new print, or at least have a dialog or something to handle this. I had 2.4 gig free which it filled up.

#### What happened instead?
```
2017-07-28 14:40:13,797 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/amazing_print_12_20170728143122-23.jpg from http://127.0.0.1:8080/?action=snapshot
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/timelapse.py", line 566, in _perform_capture
    f.flush()
IOError: [Errno 28] No space left on device
```

But then the tmp files remain on the disk, leaving it full.

I know there are several ways around this. But since timelapes can reach gigs pretty easily It suggest some kind of solution should be in order. A lazy option would be a "clear cache plugin" which cleans stuff when the distro is close to being full.

TTN-, 2017-10-21:
if the tmp folder you're talking about is /tmp it does get cleared on reboot.

guysoft, 2017-10-21:
``~/.octoprint/timelapae/tmp`` .
Dude I know linux ;)