OctoPrint/OctoPrint #2045: Incorrect printing_area from gcodeInterpreter reported with imported gcode

paukstelis, 2017-07-30:
#### What were you doing?

Using imported gcode produced with Simplify3D. This contains an initial priming line from X0 to X200.

#### What did you expect to happen?

GCODE viewer/interpreter would provide proper dimensions of the printed object size.

#### What happened instead?

The GCODE viewer/interpreter includes priming line value in the object print size.

#### Did the same happen when running OctoPrint in safe mode?

Did not attempt

#### Branch & Commit or Version of OctoPrint

1.3.4

#### Operating System running OctoPrint

Linux

#### Other notes
Perhaps it would be possible to report printing_area/dimensions after the first layer to more accurately reflect the part size? It is possible to use a start up script in octoprint instead, but there are often important features that can be achieved by generating starting scripts in the slicer instead. 

I have read the FAQ.

Kunsi, 2017-07-30:
That's not a bug. gcode viewer works exacly as intended (it displays exactly what was printed).

I think I have seen some request for that some time ago (may be related to raft/brim printing), so "customize the 'starting' layer for gcode viewer" or something like that should be in this repository somewhere.

foosel, 2017-10-12:
The whole point of the dimension calculation is to inform about the printed size of the object and whether it even fits on the bed. Your priming line is part of what your printer does. It would be wrong to ignore it (if such a thing was even possible without introducing other issues or being highly unreliable). 

I don't see why you consider this behaviour wrong in the first place?

paukstelis, 2017-10-12:
I don't consider it wrong, just not as flexible when it comes to the definition of the printed part. Original Prusa i3 by default does priming lines outside of the "printing area", and newer Marlin has definition for bed size vs. printing area, where things that might be considered printing moves (but do not contribute to the part) can be done off the printing area (priming, wipes, etc.). My motivation for exploring this was development of a plugin that would substitute initial G29 (abl) with more specific G29 to probe the bed just around the object being printed. To do that, I had to get part dimensions that excluded priming. A work around was to recalculate printing_area starting at layer 3.

foosel, 2017-10-26:
I've turned this into a request.

closed 2020-12-10 by paukstelis