OctoPrint/OctoPrint #2046: Random serial disconnecting

Project-J2, 2017-07-30:
Hi everybody,

about a few days ago, I have all day when printing is finished a random serial disconnecting : 

> State: Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1827

On my terminal I can see this : 
```
Send: M105
Recv: ok T:23.98 /0.00 B:30.28 /0.00 @:0 B@:0
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1827Please see https://bit.ly/octoserial for possible reasons of this.Changing monitoring state from 'Operational' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1827'Connection closed, closing down monitor
```
Or all Octoprint Sever crash :

> The OctoPrint server is currently not running
> 
> If you just started up your Raspberry Pi, please wait a couple of seconds, then try to refresh this page.
> 
> If the issue persists, please log into your Raspberry Pi via SSH and check the following:
> 
> Verify that the process is running: ps -ef | grep -i octoprint should show a python process.
> If it isn't, the question is why. Take a look into ~/.octoprint/logs/octoprint.log, there might be an error logged in there that helps to determine what's wrong.
> You might also want to try if you can restart the server (if no obvious error is visible): sudo service octoprint restart.
> If all that doesn't help to trouble shoot the issue, you can seek support on the OctoPrint Mailinglist or in the OctoPrint G+ Community.
> 

I have exactly the same problem on my 4 raspberry pi 3 with only Octoprint installed.

Do you know what about this few day ago? A bad update (raspberry pi or octoprint)? 
I never had this problem before.

Thanks all for your support! 👍 

nb. I give you my octoprint log : [octoprint.log.2017-07-29_secure.txt](https://github.com/foosel/OctoPrint/files/1185977/octoprint.log.2017-07-29_secure.txt)

Kunsi, 2017-07-31:
Please provide the logs requested in the issue template (octoprint.log and serial.log), so we can investigate that problem further.

Project-J2, 2017-08-05:
Hi,

I just download my serial log but it's big (more than 450Mo), I just make a screenshot of the end of the serial log : 
[image]

Can you explain how cut a part of it?

Thanks !

markwal, 2017-08-06:
It's just a text file. You can copy and paste from it to hastebin

Project-J2, 2017-08-11:
Hi,

thanks for your answer, but with this bug, I can't cut this "text" log file without loose data... I absolutely not understand why, so I attach here the file: [link](https://project-j2.box.com/s/07nxe1mulmtjucfimcdf1zn6di6174wb)

markwal, 2017-08-13:
Is it possible that the problem is in the gcode file?  (IE it got corrupted?)  It looks like it's trying to send a bunch of garbage to the printer at the end.  It *could* be just noise on the line, but if it is predictably at the end it seems like perhaps there is a problem with either the end gcode or the file is getting corrupted on save or upload.

Project-J2, 2017-08-13:
Hi @markwal , thanks for your reply!

We have checked and tried with just the end of use gcode (generated with simplify3d) without a problem.

We can upload a gcode example if you want.

markwal, 2017-08-14:
Seems like now it is either a corruption of the gcode file that caused you trouble, or you are having electromagnetic noise problems in serial communication.

You should delete serial.log though.  450Mb is getting to where you could be having i/o problems.

closed 2017-08-14 by GitIssueBot