OctoPrint/OctoPrint #2047: Object doesn't fit print volume

Cyphen-Cyn, 2017-07-30:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
aka the Contribution Guidelines that are linked ^-- just up there
in the big yellow box.

Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ...", "I have a
question ..."), that belongs on the mailing list or the G+ community
(both linked in the "guidelines for contributing" linked above, read
them!), NOT here.

Mark requests with a "[Request]" prefix in the title please. For bug
reports fully fill out the bug reporting template (if you don't know
where to find some information - it's all described in the Contribution
Guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete ANY lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to report is still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the Contribution
Guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. ...
2. ...
3. ...

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

WarpZoneNL, 2017-07-30:
Same here!

OctoPrint: 1.3.5rc2

JohnOCFII, 2017-07-31:
Does your printer do a priming strip outside the normal printing area (the Prusa i3 MK2 does, for example.  If so, you'll want to add a custom bounding box under Settings -> Printer Profiles -> <YourProfile>.

So, for example, the MK2 goes negative on Y for the priming strip, so my custom bounding box is:

```
X Coordinates: Min: 0 Max 250
Y Coordinates: Min -4 Max 210
Z Coordinates: Min 0 Max 200
```

Cyphen-Cyn, 2017-07-31:
John, 

the prime strip seems to be fully on the print bed.

I actually checked the g-code preview in Octoprint and it shows the whole print offset from center by at least 30% to the bottom left. However it will print without messing up. is there a setting i am missing somewhere or is this a bug?

foosel, 2017-08-21:
@Cyphen-Cyn @WarpZoneNL unless done already please provide:

  * sample GCODE file with which to reproduce (still needed from @WarpZoneNL)
  * configured printer profile data (bed width, depth, height, custom bounding box values, extruder count & offsets) (still needed from @Cyphen-Cyn and @WarpZoneNL)

ogoudge, 2017-08-25:
I am having this same issue with larger prints but still well with in the bed volume. I have found that if I upload the gcode and print I have no issues. However if I do anything else then come back to it I get this error. The work around has been to delete the file and re-upload it and print it right away. This has happen on more then one file now.

ogoudge, 2017-08-29:
As an update! I have to printers setup with Octoprint both are doing the same exact thing.

WarpZoneNL, 2017-08-31:
@foosel 

Sorry for my late reply, holliday ;-) 

[image]
[small_bed.gcode.txt](https://github.com/foosel/OctoPrint/files/1266680/small_bed.gcode.txt)
[image]

foosel, 2017-10-19:
@WarpZoneNL I tried to reproduce this with your file against 1.3.5 and a printer profile for a MK2 with properly configured custom bounding box to compensate for the out-of-bed priming - I can't. No warning, everything detected fine. I also noticed that your printer profile editor still seems to be all on one page - that hasn't been the case since OctoPrint 1.3.0rc1 actually, so this looks like you are running a fairly old 1.2.x version there and chances are high the issue you are experiencing has already been fixed for some time. Please check if you can reproduce it under 1.3.5.

@Cyphen-Cyn @ogoudge I can't look into this further without
  * a file with which to reproduce this (still needed from @ogoudge)
  * configured printer profile data (bed width, depth, height, custom bounding box values, extruder count & offsets - **all** of that, not just bits please) (still needed from @Cyphen-Cyn and @ogoudge)

closed 2017-11-23 by foosel