OctoPrint/OctoPrint #2050: [Feature request] Ignore wrong M105 (temperature) feedback

mrcg, 2017-08-01:
Due to sporadically bad (wrong) M105 feedback the octoprint temperature statistic is negatively impacted (peaks). The peaks do not exist in reality and the root cause are sporadic wrongly received temperature information (feedback on M105). 

In order to get rid of this i would like to ask for (optional) option to skip/ignore wrong temperature information.

Good Case:
Recv: ok T:229.5 /230.0 B:59.9 /60.0 T0:229.5 /230.0 T1:38.2 /0.0 @:81 B@:39

Bad Case (should be ignored from octoprint):
Recv: T:0.1 /230.0 B:60.0 /60.0 T0:230.1 /230.0 T1:38.2 /0.0 @:71 B@:26
Recv: T:0. /0.0 B:./.T:0. /230.0 T1:38.3 /0.0 @:66 B@:21

The rule to skipp the wrong temp feedback could be:
Skip/Ignore temp settings if ...
1. "IS" temperature from tools (T0, T1, T2, ...) or bed (B) is 0.x
2. "IS" temperature does not contain .... NUMBER + DECIMAL SEPERATION + NUMBER

This would help to have always correct statistics without any other negative impact.

I have also reported the issue at the manufacture of my printer.
Anyhow the issue is marked as "wontfix" and i am also not sure if this could be fixed on printer side at all.
The issue seems to be not specific so my printer because i know also from user people with different printer that they have this issue.

https://github.com/BCN3D/BCN3DSigma-Firmware/issues/77

foosel, 2017-08-21:
Argh... This definitely is something that would need a fix from the manufacturer, but if they refuse to do that ☹️ I can see how those wrong peaks in the graph can get annoying quickly.

This is not something I want to "fix" in OctoPrint however - as a general rule of thumb OctoPrint does not do any kind of sanity checking on the values from the printer and just displays what the printer sends because it's the authority on the temperature values and who is the octopus to judge.

What could be done however is either creating a plugin that does some preliminary parsing on received temperature lines in order to strip the corrupted ones via [the `octoprint.comm.protocol.gcode.received` hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-received) or - probably more performant - add a new hook that allows preprocessing received temperature values e.g. for sanity checking through a plugin before they are added.

foosel, 2017-08-22:
So, OctoPrint 1.3.6 will add a new hook `octoprint.comm.protocol.temperatures.received` which allows preprocessing/sanitization of the raw (but already parsed and canonicalized) temperature data as received from the printer. This allows stuff like [this little plugin](https://github.com/OctoPrint/Plugin-Examples/blob/master/sanitize_temperatures.py) to filter out data that's outside of "sane" ranges.

Change is pushed to `maintenance`, soon `devel`, will go into 1.3.6.

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel