OctoPrint/OctoPrint #2051: [ Request]

Chupaflo, 2017-08-01:
Hi, I thought it could be nice to have a way to watch the timelapses on the web interface. Just a simple video player. Just an idea =)

Kunsi, 2017-08-02:
Your feature Request is missing a title (and if you remove the space after the '[', the bot will no longer complain).

"Watching Timelapses in GUI" also looks like something that could be done using a plugin :)

Chupaflo, 2017-08-02:
Thanks for your quick answer =)

foosel, 2017-08-21:
Problem is that the timelapses are mpg (thanks to a cross platform way to encode mp4 on stuff like a Pi being hard to realize) and that isn't natively supported as playback format in the browser. Otherwise I'd probably have added a plugin for that already ;) But maybe someone sees that and has an idea on how to get it to work after all :)

foosel, 2018-05-08:
webm has the same problems with regards to cross platform way of (fast) encoding as mp4.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel