OctoPrint/OctoPrint #2052: [Request] File mask customization for upload dialog

TheLongRunSmoke, 2017-08-03:
I think, it'd be good to allow user to customize, what kind of files he want to see in upload dialog.
Sorry, if I inattentively read FAQ andr YAML instruction.

markwal, 2017-08-03:
By default the upload dialog shows the files that the uploader will allow to be uploaded (this is extensible via plugin).  In Chrome on Windows, you can also choose "All Files" to see everything, but it still will only allow upload of allowed file types.

What kinda files were you hoping to upload?

TheLongRunSmoke, 2017-08-04:
I want to see *.gcode only. 
Is extensible via plugin? Then I just write the plugin.

markwal, 2017-08-04:
Well, it's extensible where you can add file types.  I don't think you can remove them.  So you don't want it to show .stl and .gco I guess.

TheLongRunSmoke, 2017-08-06:
Wrote the plugin. It's a little rough, I just replace value in button's "accept" field. Not done with bindings yet, and not sure that's working for everyone, but [code available there](https://github.com/TheLongRunSmoke/OctoPrint-Filetypes), from dev branch.
Any suggestions?

foosel, 2017-08-23:
Closing this since @TheLongRunSmoke wrote a plugin to solve this issue themselves, which has now been [published on the repository](http://plugins.octoprint.org/plugins/filetypes/) :)

closed 2017-08-23 by foosel