OctoPrint/OctoPrint #2053: Estimated Total Print Time on GCode Viewer tab incorrect

jrucker2004, 2017-08-04:
#### What were you doing?
About 3/4 of the way through a print, I was looking at the GCode Viewer tab for the first time, and noticed the time estimate was double the time estimate on the left column, in the "state" section

#### What did you expect to happen?
I expected the time estimates to be consistent across the app

#### What happened instead?
It appears that either a math error or a wildly different time estimation algorithm is being used.

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
Monoprice maker select V2, unknown firmware, purchased July 2017

#### Browser and Version of Browser, Operating System running Browser
Chrome 59.0.3071.115
Windows 7

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
[image]

markwal, 2017-08-04:
Yep.  The gcode viewer computes the total print time using an algorithm that runs in javascript entirely in the browser.  The other time estimate comes from the server and actually uses several different algorithms depending on what information it has available.  In particular, as you get closer to finished OctoPrint weights the bytes remaining to be consumed method of computation as more accurate, but the gcode viewer just happily keeps showing its original analysis.

There are quite a few discussions on the topic of estimated time.  Also, foosel discusses the topic in one of the patreon videos.  I think it was this one: OctoPrint On Air 3 https://www.youtube.com/watch?v=nCAMjqysBB0

markwal, 2017-08-04:
The link to the 44 minute point where the discussion starts: https://youtu.be/nCAMjqysBB0?t=44m45s

fiveangle, 2017-08-04:
Gina, just have issuebot look for "print estimate" and suggest watching that link ;)

closed 2017-08-19 by GitIssueBot