OctoPrint/OctoPrint #2054: [Request] choose filament slicing

leobel96, 2017-08-04:
Hello. Is there a way to choose filament for slicing? I saw that in Settings I can add filaments but this is only for heatbed and nozzle preheating and not also for printing. Should I upload a different printing profile for EVERY filament I use? Thank you

Kunsi, 2017-08-08:
Slicing is provided by either [https://github.com/OctoPrint/OctoPrint-Slic3r/issues Slic3r plugin] or [https://github.com/OctoPrint/OctoPrint-CuraEngine/issues Cura plugin]. So that request is related to whichever plugin you're using. Please refer to the issue tracker of the corresponding plugin for your request.

leobel96, 2017-08-08:
@Kunsi The point is that, how you can see in this screen, there isn't a selector for filament. Regardless of the slicer chosen, the screen doesn't change and the selectable items remain that same. So I think it's a limitation of octoprint and not of the slicer.

 [image]

foosel, 2017-08-21:
> Is there a way to choose filament for slicing

No, this is simply not part of the (limited) slicing workflow in OctoPrint and also not a priority. Some slicers allow stuff like filament profiles, others don't, so it's not trivial to solve either. The current approach is to define one slicer profile per filament - or to use your desktop slicer and just upload the file to OctoPrint (Cura & Slic3r even support doing this automatically).

For anything more sophisticated than that I suggest to take a look at [e.g. this plugin](http://plugins.octoprint.org/plugins/slicer/)-

closed 2017-08-21 by foosel