OctoPrint/OctoPrint #2055: System API field too small for value

b-morgan, 2017-08-04:
Lulzbot Cura 2 beta has the ability to connect to OctoPrint but requires an API key. When I go to OctoPrint settings, API the key is shown, but the field is too small. When I mouseover the field, I get a "red circle with diagonal line" cursor. This appears to be independent of browser as I tried Chrome, Firefox, and Edge.

[image]

b-morgan, 2017-08-04:
#### What were you doing?

Lulzbot Cura 2 beta has the ability to connect to OctoPrint but requires an API key. When I go to OctoPrint settings, API the key is shown, but the field is too small. When I mouseover the field, I get a "red circle with diagonal line" cursor. This appears to be independent of browser as I tried Chrome, Firefox, and Edge.#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

1.3.4

#### Operating System running OctoPrint

OctoPi, version 0.14.0 

#### Printer model & used firmware incl. version

Lulzbot TAZ 6, RAMBo 1.3L, Marlin 1.0.2.22

#### Browser and Version of Browser, Operating System running Browser

Windows 10 Professional, Chrome 60.0.3112.90 (Official Build) (64-bit), Firefox 54.0.1 (64-bit)

#### Link to octoprint.log

[octoprint.zip](https://github.com/foosel/OctoPrint/files/1200672/octoprint.zip)

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

Yes, I have read the FAQ.

KC703, 2017-08-04:
Start the mouse above the API Key box and drag down... the API key should
be highlighted for cut and paste...

foosel, 2017-08-22:
I've increased the size of the field and added a little "copy" button (see also #2056). 

[image]

Already on `maintenance`, soon on `devel`, will be released with 1.3.6.

commit e9401b70 (2017-08-22): Easier copying of API keys

There's now a dedicated button for that.

This should get around issues like #2056. Also increased size
of API key display field (where possible). Fixes #2055
closed 2017-12-12 by foosel