OctoPrint/OctoPrint #2056: System API key cannot be copied

b-morgan, 2017-08-04:
When attempting to "copy" the API key, nothing happens in Chrome and Firefox but it works with Edge. I do have LastPass (but Edge doesn't support it).

b-morgan, 2017-08-04:
What were you doing?

When attempting to "copy" the API key, nothing happens in Chrome and Firefox but it works with Edge. I do have LastPass (but Edge doesn't support it).

#### What did you expect to happen?

Key in clipboard.

#### What happened instead?

Nothing.

Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

Branch & Commit or Version of OctoPrint

1.3.4

Operating System running OctoPrint

OctoPi, version 0.14.0

Printer model & used firmware incl. version

Lulzbot TAZ 6, RAMBo 1.3L, Marlin 1.0.2.22

Browser and Version of Browser, Operating System running Browser

Windows 10 Professional, Chrome 60.0.3112.90 (Official Build) (64-bit), Firefox 54.0.1 (64-bit)

Link to octoprint.log

[octoprint.zip](https://github.com/foosel/OctoPrint/files/1200711/octoprint.zip)

markwal, 2017-08-04:
I think this is a security fix it some browsers.  Some webapps (OneNote comes to mind) pop up a message when you click the copy button that says you have to use Ctrl-C or the right click menu.

b-morgan, 2017-08-04:
I tried ctrl-C and right-click Copy. Neither worked (in Chrome and Firefox). Both worked in Edge.

ntoff, 2017-08-04:
Works ok here on chrome running on windows 10 pro

double click the key (or click and drag) to highlight it all  -> right click -> copy. Does that not work for you?

b-morgan, 2017-08-05:
To reiterate what I said in my previous posts, on Windows 10 Pro, Chrome 60.0.3112 I can highlight the key but right-click Copy does not work for me. Same results using Firefox (right-click Copy doesn't work).

With Microsoft's Edge browser, I can highlight and right-click Copy without any issues.

I think a key piece of this puzzle is that I have LastPass installed which has addons for Chrome and Firefox but does not have one for Edge (or rather, Edge doesn't yet support addons). I did try disabling LastPass in Firefox but that did NOT change the symptoms (right-click Copy doesn't work).

ntoff, 2017-08-05:
Ok well it works in chrome but I just tried firefox and even without lastpass I can't copy the key either. What I CAN do, is drag it onto notepad++ which inserts it into a document and then leaves OctoPrint hanging in limbo on the upload overlay so I guess bug confirmed? But only for firefox. Chrome still works ok for me (I don't use lastpass).

KC703, 2017-08-05:
Right clicking in both Chrome (Version 60.0.3112.90 (Official Build)
(64-bit)) and Edge allow me to copy the API Key...  Ctrl+C works also when
the key is highlighted.

I use the Lastpass plugin for both Chrome and Edge (not sure if this is a
plugin, could be the Lastpass app from the MS Store).

Just tried Fiefox and the API key won't highlight...

b-morgan, 2017-08-05:
I'm a little embarrassed. I just tried Chrome again and it works. That leaves us with a Firefox issue which you all have confirmed.

ntoff, 2017-08-11:
I wonder if a little "copy" button should be placed next to the API key, allowing someone to just press that to copy the key into the clipboard.

foosel, 2017-08-22:
I have no idea what Firefox' problem is here, but regardless, I've added a bunch of copy buttons to the various API key fields (Settings > API, Settings > Access Control, User settings).

[image]

Pushed to `maintenance`, soon `devel` and to be released with 1.3.6.

sepal, 2017-12-12:
Just had this issue as well and did a little bit of research:
Apperently this is new in Firefox Quantum (aka 57.0) and is a [bug fix](https://support.mozilla.org/de/questions/1190785), since the HTML5 standard specifies that users shouldn't be able to interact with a disabled inputs.
The API key input should instead use the [read only attribute](https://developer.mozilla.org/de/docs/Web/HTML/Element/Input#attr-readonly), which allows to copy from, but not manipulate the form elements.

foosel, 2017-12-12:
> The API key input should instead use the [read only attribute](https://developer.mozilla.org/de/docs/Web/HTML/Element/Input#attr-readonly), which allows to copy from, but not manipulate the form elements.

And that was changed in cf25291 and will be part of 1.3.6 as well.

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel