OctoPrint/OctoPrint #2060: multiple instance

TylonHH, 2017-08-06:
#### What were you doing?
Nothing special. I tried some RC but now I'm on  

#### What did you expect to happen?
Faster loading of the web UI, no interrupts while printing and upoading a stl via curl

#### What happened instead?
I have multiple instance of octoprint and mjpeg streamer
And I have still little issues with stopping a print. Please see https://github.com/foosel/OctoPrint/issues/1946#issuecomment-316700413
WebUi is loading long. And someone told me that my htop doesnt look normal

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Branch & Commit or Version of OctoPrint
1.3.5rc2

#### Operating System running OctoPrint
OctoPi 0.13.0

#### Printer model & used firmware incl. version
CoreXY Marlin RC 8

#### Browser and Version of Browser, Operating System running Browser
Firefox 54.0.1 (64-Bit)

#### Link to octoprint.log
https://pastebin.com/MGkmQZaY

#### Link to contents of terminal tab or serial.log
log is too big (5MB)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2017-08-21:
I'm sorry but what specifically is this ticket about?

foosel, 2017-10-02:
Closing because I'm still completely unable to figure out what exactly the problem is you are reporting here. Feel free to request a reopen when you can clarify that.

> Multiple instance

That's not very descriptive as a title and I have no idea what you mean by that. 

> WebUi is loading long

Define "long". I'm seeing ~5s here against a Pi2 (longer after initial startup before it is cached). Loading speed heavily depends on network quality and installed plugins as well as OctoPrint itself, so it's pretty much impossible to debug this remotely with this amount of information.

> And I have still little issues with stopping a print. Please see #1946 (comment)
 
If your problem is still the same as in the ticket you linked, then by all means please try what was suggested in there first, then report back *in there*. It doesn't help creating multiple tickets for identical issues, in fact it makes things even trickier to take care of.

> And someone told me that my htop doesnt look normal

To me it looks normal. htop shows threads by default, see also [here](https://unix.stackexchange.com/a/10403), which explains why you are seeing multiple entries there. Press `H` to hide threads. Press `F5` to get a tree view of things that makes stuff clearer.

closed 2017-10-02 by foosel