OctoPrint/OctoPrint #2062: Nozzle Offset Misrepresented in GCode Viewer

AcrimoniousMirth, 2017-08-08:
#### What were you doing?

I was doing a dual extrusion test print. STL sliced with Simplify 3D v.4.0. Running Marlin as firmware. The issue pertains to the dual nozzle offset, i.e. X-Y distance that nozzle 2 is away from nozzle 1. These were the same in both Simplify 3D and the Firmware. The test print was to check nozzle alignment and emurenthe values were the same. Print was simply a cube printed with the second nozzle on top of a cube printed by the first. The offset in X and Y is found with callipers, averaged and new values tried.

#### What did you expect to happen?

I expected, when it came to switching nozzles, that Octoprint would display the “true” print whereby the GCode viewer would show the second cube being printed on top of the first.

#### What happened instead?

The second cube was offset from the first by the X-Y values of the nozzle offset. Physical print was without error but the code representation was incorrect.

#### Did the same happen when running OctoPrint in safe mode?

Could not tell you but I doubt it would be any different. 

#### Branch & Commit or Version of OctoPrint

1.3.4 Master

#### Operating System running OctoPrint

MacOS Sierra 10.12.1 Beta (16B23e8c) 

#### Printer model & used firmware incl. version

own-design CoreXY, Marlin RC-7

#### Browser and Version of Browser, Operating System running Browser

Safari, checked on both OS and iOS. Issue is not with computer or browser.

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

Sorry, this occurred late last night and I didn’t think to grab a screenshot. Will try and upload a screengrab shortly.

I have read the FAQ.

AcrimoniousMirth, 2017-08-08:
[Simplify 3D error Footage.mov.zip](https://github.com/foosel/OctoPrint/files/1207573/Simplify.3D.error.Footage.mov.zip)

FHeilmann, 2017-08-09:
Did you configure the nozzle offset in Octoprint, too? You need to tell Octoprint about the displacement of the nozzles. S3D corrects for the offset, which results in G-Code that looks spaced apart to Octoprint because Octoprint doesn't know about the displacement.

So not really a bug, but rather a configuration issue.

AcrimoniousMirth, 2017-08-09:
No, the nozzles are correctly configured in Octoprint. Octoprint **is** aware of the displacement.

FHeilmann, 2017-08-09:
Can you clarify for me where you configured the offsets (as in really punched the numbers in) in the following entities:

S3D
Printer Firmware
Octoprint

So I can understand your problem better.

AcrimoniousMirth, 2017-08-09:
Certainly.
I manually configured (input the numbers into):
1) Simplify 3D (Processes→G-Code→Update Machine Definition→Toolhead offsets→Tool 1); X=19.54, Y=11.25
2) Marlin (configuration.h); X=19.54, Y=11.25
3) Octoprint (Settings→Printer Profiles→[Printer Name]→Hotend & Extruder→ Offset T1); X=19.54, Y=11.25

So they are correctly configured across the firmware, slicer and octoprint.

FHeilmann, 2017-08-09:
What really surprises me is that your print is that your print is going well. Because if you configure the offset in both S3D AND Marlin your two colors shouldn't come out aligned because 

S3D modifies the coordinates 

(Say T0 stops printing at X100 Y100 then, after switching to T1, S3D will generate G-Code that starts printing at X80.46 Y88,75 to account for the offset)

and Marlin internally corrects for the offset

If the 2nd nozzle is active, G-Code for position X80.46 Y88,75 would be internally corrected to X60 Y77.50)

Usually you only either tell S3D about the offset (and octoprint so it can "undo" the offset when drawing the G-Code) or you configure Marlin and don't tell octoprint about it.

AcrimoniousMirth, 2017-08-09:
Yes, I wasn’t entirely sure at first as to how it would handle it. Inputting the offset into both seemed the logical thing to do and if it failed would inform me. I assume that either/or just has a function that tests for the setting and overrides.

foosel, 2017-08-21:
While I tend to agree that somewhere in that work flow the offset probably gets applied too often ;) if it prints right it sounds like also should display correctly. So, @AcrimoniousMirth, could you please provide a GCODE file with this to test this?

AcrimoniousMirth, 2017-08-21:
Hi, thanks for the reply, and also thank you for all the work you've done, I've been an avid user of Octoprint for a long time now and hopefully someday my C++ skills will be good enough that I can contribute :)
I think that setting the Octoprint offset to 0,0 would work, but that's not strictly true and if that's the solution then a little explanation of what to do and why in that case could be added in.
I'm attaching the file here, it's similar to the one displayed in the video but hollowed out. Bottom cube is printed with Tool 0, head parks at machine front to cool down Tool 1 and heat up Tool 2 then resumes print with Tool 2.
Please note, I use an origin-centred bed so will most likely show at the bottom corner for you.

[Nozzle Alignment Cubes (41min).gcode.zip](https://github.com/foosel/OctoPrint/files/1238771/Nozzle.Alignment.Cubes.41min.gcode.zip)

foosel, 2017-10-19:
So thanks to the Flu Of Doom that took a bit longer than anticipated.

I took a look into your GCODE file and funnily enough even though the profile data on top shows that you configured an offset, it doesn't reflect in the actual generated GCODE:

```
; outer perimeter
G1 X9.760 Y-9.760 F4800
G1 X9.760 Y9.760 E0.9739 F698
G1 X-9.760 Y9.760 E0.9739
G1 X-9.760 Y-9.760 E0.9739
G1 X9.560 Y-9.760 E0.9639
G1 X9.760 Y-9.760 F698

[...]

T1

[...]

; outer perimeter
G1 X-9.760 Y-9.760 F4800
G1 X9.760 Y-9.760 E0.9739 F698
G1 X9.760 Y9.760 E0.9739
G1 X-9.760 Y9.760 E0.9739
G1 X-9.760 Y-9.360 E0.9539
G1 X-9.760 Y-9.760 F698
G1 E-3.5000 F1800
```

The first bit is the outer perimeter for T0. Then the change to T1 comes. Then the outer perimeter coordinates for T1. Notice how they are the same. With a configured offset I'd expected S3D to generate offset coordinates here (and hence cause havoc since that offset is also already configured in your firmware - as mentioned, you do one **or** the other).

So something in your slicer profile tells S3D "There's an offset here, but just ignore it" for some reason and that makes stuff work fine for you. However, OctoPrint DOES reverse-apply said offset that you configured since it expects the GCODE it receives to have it applied already (and that would cause issues in the viewer). And since your GCODE actually is offset-free, you get rendering issues. The correct approach here is to set a zero offset in OctoPrint as well, since your GCODE (for reasons I do not understand though) is actually offset free.

foosel, 2017-10-19:
Closing this since there's nothing to do from OctoPrint's side here - it's actually behaving correctly, even though that's unwanted here ;)

closed 2017-10-19 by foosel