OctoPrint/OctoPrint #2065: I cannot get my webcam to flip.

zq8blazer, 2017-08-10:
I have a Logitech C270 mounted upside down and the latest version of Octoprint on a Raspberry Pi 3.  I have checked flip vertically, flip horizontally, saved, and rebooted Octoprint.  The camera will not rotate and I was wondering what I can do to fix this issue?

[image]
[image]

JohnOCFII, 2017-08-10:
> The camera will not rotate and I was wondering what I can do to fix this issue?

Try the -rot feature on the ```camera_usb_options=" "``` line in your octopi.txt file in /boot.

See this for more info:

https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration

zq8blazer, 2017-08-10:
Do I go to terminal in Octoprint and enter -rot camera_usb_options="180" ?

zq8blazer, 2017-08-11:
This is how I did it; however, the video feed still isn't flipped.[webcam 3](https://user-images.githubusercontent.com/25054144/29198303-503247e4-7e08-11e7-9b2f-d46a4e408532.png)

zq8blazer, 2017-08-11:
This is how I did it; however, the video feed still isn't flipped. 
[image]

JohnOCFII, 2017-08-11:
Not sure how many times you can set that option, and there is already one line with ```camera_usb_options``` in that file.  I'd suggest you combine them to:

```camera_usb_options="-r 1280x720 -f 30 -rot 180"```

(I haven't actually needed this myself, but looking at the documentation, it seems like the right way to go.  Be sure to re-start the Pi after you've made this change to octopi.txt)

zq8blazer, 2017-08-11:
Thanks, but that didn't work either :(

zq8blazer, 2017-08-11:
I just noticed that when I'm at the /webcam/?action=stream page it hasn't flipped 180 degrees, but on the Octoprint screen under the control tab the image is flipped as it should be.

ntoff, 2017-08-11:
The feed from the webcam  (/webcam/?action=stream) can't be flipped by octoprint. Octoprint only has control over the image it displays on the control page, and it flips it just by flipping the image your browser displays, not the actual source image. If you view the stream directly, you're viewing what comes out of mjpeg streamer, octoprint has no control over that stream.

https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration according to this page, -rot isn't supported by the usb camera options (input uvc), it's only supported by the raspberry pi input plugin for mjpeg streamer.

zq8blazer, 2017-08-11:
okay thank you.  So just to clarify there is no way to do this?

ntoff, 2017-08-12:
See @foosel 's response here: https://github.com/guysoft/OctoPi/issues/369 (foosel owns octoprint)

and just to be clear, mjpg-streamer is **not** part of octoprint, it is a 3rd party program. You can use any webcam server that outputs an mjpeg stream, so it's up to the streaming server as to which options it supports in terms of rotations, flipping the image, etc.

foosel, 2017-08-21:
For the record, `-rot` is a setting that the input driver for the raspicam allows, the uvc (=USB camera) input driver does NOT support it.

If I understand this ticket correctly, the flip/rotate settings in OctoPrint aren't the problem here and work just fine, it's the mjpg-stream itself that stays as is - which as already pointed out is to be expected and nothing OctoPrint can fix. Closing this therefore. If I misunderstood the problem on hand, feel free to clarify and request a reopen.

closed 2017-08-21 by foosel

sphereinabox, 2019-11-11:
setting camera_usb_options to include `-hf true -vf true` to both flip horizontally and vertically is another way to rotate 180 the camera view that is supported by the v4l driver with my ps3 eye toy