OctoPrint/OctoPrint #2066: [Brainstorming]  Python Code cleanup/linting

Cabalist, 2017-08-11:
Is there any interest in some pull requests with some maintenance work on issues a linter finds?  This is stuff like identifying staticmethods, fixing name collisions with built-ins, broad exception catches, etc.

I'd like to get some familiarity with the code base and reading through it with my linter on is helpful for that.    The code in general looks very very clean.  At the moment I don't even know what branch to target for something like this.  :)

Not hurt feelings if this isn't something that is a priority at the moment.

ntoff, 2017-08-11:
>At the moment I don't even know what branch to target for something like this. 

The contribution guidelines https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md say either maintenance or devel for adding new features.

>Create your pull request only against the maintenance or devel branch

I'd say you should probably check out one of those branches, then create a new branch from that like devel/lintedcleanup or something.

Cabalist, 2017-08-11:
Cool.  Will do.  I'll start with devel.  Thanks for the guidance.

foosel, 2017-08-21:
Any help with that is greatly appreciated. Just be aware that if a change turns out to be very disruptive (e.g. leads to most of a file having to change or something like this) it might cause some back and forth, especially if backwards compatibility might be impacted. However, from what you write about your plans, I think that should hopefully not even be the case :)

LazeMSS, 2021-03-31:
@foosel  is this not implemented as part of the CI/CD?

foosel, 2021-03-31:
Indeed! We have flake running now, with some plugins. Plus some more checks.

closed 2021-03-31 by foosel