OctoPrint/OctoPrint #2067: [Request] Octoprint Timelapse

ComicBit, 2017-08-11:
Hello, i just found a crashed print because of Z Based timelapse on Vase Mode. Essentially Octoprint triggers the capture on Z-Changing but this thing obviusly triggers continuosly on Vase Mode, it can corrupt a print (as happened to me just now). I know that the simpler solution is changing the mode but isn't possible to make an automatic system that recognize a vase mode and prompt a proposal of changing the Timelapse Mode?

FHeilmann, 2017-08-11:
The page where you configure timelapses EXPLICITLY states to not use it for vases.

How would octoprint detect a vase mode from G-Code?

ComicBit, 2017-08-11:
Yes i know, but you can ocasionally forget it.. For example Simplify3D write this on the gcode spiralVaseMode,1

Octoprint will possibly search for this strings

ntoff, 2017-08-12:
Not every slicer puts that info there though, some might not put any info at all so it'd be a feature that works sometimes, maybe, but only under very certain circumstances.

foosel, 2017-08-21:
OctoPrint can't search for these strings since

  * they are not standardized in any way
  * in order to be aware of them OctoPrint first has to read the file fully - this takes time and CPU resources and if you hit "Print" before that is done, you are back to square one

What could be done though is rate-limiting the z-change triggers, basically not doing another snapshot if the last one was less than n seconds ago. The tricky bit will be to find a good default value here, but that should be do-able (and could also be made configurable to accommodate corner cases)

foosel, 2017-10-26:
Rate limiting is now available on `maintenance`. The default minimum delay between snapshots is 5s but this can also be changed via the configuration dialog:

[image]

The minimum is 1s. 

Will be released with 1.3.6.

ComicBit, 2017-10-26:
As always an awesome solution, thanks for your job!

closed 2017-10-26 by ComicBit

ntoff, 2017-10-26:
What's "reset to active configuration"? Is that like "reload configuration from disk" if you've changed something and want to go back to the way it's set in config.yaml?

foosel, 2017-10-27:
Basically, "load whatever's the currently active configuration". That doesn't necessarily need to be what's in `config.yaml`, since there it will only be stored if you "save as default", but rather what the server has currently set as active timelapse configuration in memory. In a nutshell, click that button and it reverts to what the form was showing when you loaded the page or when you last clicked "save changes".