OctoPrint/OctoPrint #2069: [Request] New Interface

ComicBit, 2017-08-11:
OctoPrint is starting to looks old to me compared to other things. It isn't responsive and in general the layout seems to be not that great, i would work on it, i'm a web developer if needed with mobile first and HTML 5 in mind :)

FHeilmann, 2017-08-11:
Octoprint supports creating new interfaces via plugins.

One example for this is the touchUI plugin which can be found here:
http://plugins.octoprint.org/plugins/touchui/

If you don't like the current layout, feel free to create a new one as a plugin so people can choose whether they want yours or the default one.

JohnOCFII, 2017-08-11:
Also - here are some recent comments on the level of effort required for UI changes: https://github.com/foosel/OctoPrint/issues/1999

closed 2017-08-11 by ComicBit