OctoPrint/OctoPrint #2070: [Request] Gcode patterns shortcut

ComicBit, 2017-08-11:
I think would be nice to have a window where you can put a series of instructions and give them a name, for example if i want to clean the nozzle i would create a macro and call it "Clean Nozzle" and put the gcode: "G12 P0 S7" and having a tiny window where this macro would be accessible i think will be awesome

FHeilmann, 2017-08-11:
this can already be done using the custom controls system. Please take the time to read and search the topics regarding your issues. Half of them already have a lot of existing work towards them or have already been discussed

https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls

Please close this issue if your request is satisfied.

ComicBit, 2017-08-11:
This is not what i was talking about. I'm already using this. But i'm talking about simple Gcode macro that you can call without opening a menu

ntoff, 2017-08-12:
If you install the custom control editor, http://plugins.octoprint.org/plugins/customControl/ , it makes adding those buttons a lot easier.

>This is not what i was talking about. I'm already using this. But i'm talking about simple Gcode macro that you can call without opening a menu

So do you mean a section of OctoPrint that is visible all the time, no matter which tab you're on? Something over in the sidebar?

foosel, 2017-08-21:
Already implemented via custom controls for which the corresponding editor plugin has also kindly been linked to as well.

See also [the docs](http://docs.octoprint.org/en/master/features/custom_controls.html).

closed 2017-08-21 by foosel