OctoPrint/OctoPrint #2071: [Request] Emergency Stop Button

ComicBit, 2017-08-11:
Seems that the most immediate mode to immediately stop wathever the printer is doing is disconnect/riconnect to the printer and M80/M81 do the job too. I think would be nice to have a emergency button on the interface

FHeilmann, 2017-08-11:
Same as my other reply.

Create a custom control to send M112 and whatever else you want to send during an emergency stop

https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls

ComicBit, 2017-08-11:
FHeilmann honestly i can't understand why your replies are so disappointing. I know that Octoprint is an open source platform and that i can program it for do what i want, but this is not the point of the "request" i have made today. They doesn't need to me, i can play with octoprint. They are just some things i think would be great to be added to the octoprint core. I know i can do a custom button to stop the machine (i have it already) but you really think that a emergency button placed underneath in a menu is a fast and reliable choice?

This said i was coming here for helping, if the only thing that i have to read in reply is "you can already done it" i will not write anything anymore. Maybe octoprint have to still as it is forever because we can customize it instead of make it better ;)

FHeilmann, 2017-08-11:
Custom controls allows you to create simple buttons that execute a G-Code Macro when pressed. Isn't this what you want? The button will be on the "Controls" tab and look just like the one (for example the "Motors off" button) I don't understand what you mean with the "underneath a menu" do you mean the System Commands editor?

The reason there is no size fits all solution is that octoprint is used with a multitude of firmwares for 3D printers and CNC machines. There is no "one G-Code Macro" to do what you want for all printer Firmwares (for example, a lot of PPL don't use an ATX supply, which makes M80/M81 useless)

Sunoo, 2017-08-11:
There is a plugin that takes care of exactly what you're asking for rather nicely: http://plugins.octoprint.org/plugins/simpleemergencystop/

ntoff, 2017-08-12:
and I have my own plugin (not registered so it needs manual install)

https://github.com/ntoff/OctoPrint-Estop

The reason I didn't register it is because all it does is send M112 and how the printer responds to that depends on the firmware. Lots of firmwares have "blocking" moves which will ignore any incoming commands until the previous moves are done. Until firmwares stop doing that and implement a proper emergency stop that works immediately for _all_ situations, a software emergency stop button is pretty useless so I prefer my hardware option (i.e turn the power off manually by hand).

foosel, 2017-08-21:
As already pointed out, there are a bunch of plugins already that provide this and the whole point of the plugin system is to allow easy extension with functionality people need without forcing that on everyone who might not have any use for it.

And the general problem with `M112` has also already been pointed out: it's more of a bandaid than a real emergency stop since it relies on the serial connection to the printer still working *and* accepting commands in the first place - good luck with that if push really comes to shove. A *proper* emergency stop implementation that users can acutally *rely* on to work would require additional hardware and that opens a whole other can of worms on something that is capable of running on as many platforms and in as many environments as OctoPrint, so is also something that would have to be solved through a plugin.

I would love to offer a big prominent emergency stop button in OctoPrint. But I'm not going to offer something like that when it doesn't work in most real emergency cases - it would only provide a false sense of security.

And I don't even want to start on the "confirmation dialog yes or now" discussion that would result from adding something like that to the core application ;)

closed 2017-08-21 by foosel