OctoPrint/OctoPrint #2072: wont start after install on pi

jvon, 2017-08-12:
Starting the server via "octoprint" is deprecated, please use "octoprint serve" from now on.
2017-08-12 03:53:47,304 - py.warnings - WARNING - /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/webassets-0.12.1-py2.7.egg/webassets/filter/__init__.py:722: UserWarning: Error while loading builtin filter module 'webassets.filter.jade': bad marshal data (string ref out of range)
  'module \'%s\': %s' % (module_name, e))

2017-08-12 03:53:48,566 - octoprint.server - INFO - ******************************************************************************
2017-08-12 03:53:48,573 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-08-12 03:53:48,579 - octoprint.server - INFO - --- Logged during platform initialization: -----------------------------------
2017-08-12 03:53:47,304 - py.warnings - WARNING - /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/webassets-0.12.1-py2.7.egg/webassets/filter/__init__.py:722: UserWarning: Error while loading builtin filter module 'webassets.filter.jade': bad marshal data (string ref out of range)
  'module \'%s\': %s' % (module_name, e))

2017-08-12 03:53:48,593 - octoprint.server - INFO - ------------------------------------------------------------------------------
2017-08-12 03:53:48,599 - octoprint.server - INFO - ******************************************************************************
2017-08-12 03:53:49,142 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-08-12 03:53:49,887 - octoprint.plugin.core - ERROR - Error loading plugin corewizard
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard/__init__.py", line 12, in <module>
    from flask.ext.babel import gettext
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 29, in <module>
    from .ctx import RequestContext, AppContext, _AppCtxGlobals
EOFError: EOF read where object expected
2017-08-12 03:53:49,955 - octoprint.plugin.core - ERROR - Error loading plugin cura
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura/__init__.py", line 11, in <module>
    import flask
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 26, in <module>
    from . import json
ImportError: cannot import name json
2017-08-12 03:53:51,053 - octoprint.plugin.core - ERROR - Error loading plugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 13, in <module>
    from octoprint.server.util.flask import restricted_access, with_revalidation_checking, check_etag
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/__init__.py", line 10, in <module>
    from flask import Flask, g, request, session, Blueprint, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 26, in <module>
    from . import json
ImportError: cannot import name json
2017-08-12 03:53:51,117 - octoprint.plugin.core - ERROR - Error loading plugin discovery
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery/__init__.py", line 14, in <module>
    import flask
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 26, in <module>
    from . import json
ImportError: cannot import name json
2017-08-12 03:53:52,332 - octoprint.plugin.core - ERROR - Error loading plugin announcements
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements/__init__.py", line 19, in <module>
    import flask
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 26, in <module>
    from . import json
ImportError: cannot import name json
2017-08-12 03:53:53,057 - octoprint.plugin.core - ERROR - Error loading plugin softwareupdate
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 11, in <module>
    import flask
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 26, in <module>
    from . import json
ImportError: cannot import name json
2017-08-12 03:53:55,444 - octoprint.plugin.core - INFO - Found 1 plugin(s) providing 1 mixin implementations, 1 hook handlers
2017-08-12 03:53:55,477 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/bin/octoprint", line 9, in <module>
    load_entry_point('OctoPrint==1.3.4', 'console_scripts', 'octoprint')()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/__init__.py", line 405, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 696, in main
    rv = self.invoke(ctx)

jvon, 2017-08-13:
#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. installed according to the install on a current raspbian install
2. pi@octopi:~ $ ~/OctoPrint/venv/bin/octoprint serve 

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen? the service start and run

#### What happened instead? bunch of errors

#### Did the same happen when running OctoPrint in safe mode? yes same thing

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please,
OctoPi's version can be found in /etc/octopi_version]

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

#### Printer model & used firmware incl. version

wanhao duplicator i3

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ. yes

foosel, 2017-08-21:
This looks like a severely corrupted install of python core libraries and some dependencies:

```
EOFError: EOF read where object expected
ImportError: cannot import name json
```

Might be caused by file system corruption (always shut down your pi fully before disconnecting power, the green LED should no longer flicker) or something else. Since it's impossible to say I'd suggest to try a reinstall, see if that magically fixes it.

foosel, 2017-09-28:
Any news on whether a reinstall fixed that?

closed 2017-10-11 by foosel