OctoPrint/OctoPrint #2076: [Request]

brettlg, 2017-08-13:
Is is possible to have multiple watched folders with 1 slicing profile assigned to 1 watched folder. i.e. watched folder "ABS 0.1mm" with a corresponding slicer profile, watched folder "PLA 0.4mm" with corresponding slicer profile. Goal different folders different profiles applied, simple drag and drop to appropriate folder for processing. Hope this is the correct place to post.

foosel, 2017-08-21:
> Is is possible to have multiple watched folders with 1 slicing profile assigned to 1 watched folder

No, currently this is not possible. Slicing also doesn't have a high priority (most people - myself included - prefer using the actual slicer and uploading the GCODE to OctoPrint).

It might however be possible to do through a plugin (which someone would need to write first).

brettlg, 2017-08-22:
Thanks for your reply. 
I am also using slic3r straight to Octoprint but hope eventually to get profiles fine tuned enough to just drag and drop to a hot folder for processing.

I love Octoprint.  Wouldn’t bother 3d printing without it.