OctoPrint/OctoPrint #2077: Issue with octoprint in a single nozzle/heater dual bowden extruder with a Y splitter

cristiansavaro, 2017-08-13:
It seems that octoprint (1.3.4) doesn't fully support my setup. I am running Marlin 1.1.4 on my 3d printer.
I can print with each "toolhead" separately but when doing so with the second tool (T1) the temperatures 
aren't displayed correctly. I am attaching a few screen captures with my issue (highlighted in yellow).

[image]
[image]
[image]

cristiansavaro, 2017-08-13:
#### What were you doing?

Trying to print in a single nozzle/heater dual bowden extruder with a Y splitter where the toolhead in terms of a heater is refered to as T0 but the bowden extruders are both T0 and T1.

#### What did you expect to happen?

I expected the temperatures to be correctly reported and graphed.

#### What happened instead?

As soon as toolhead is changed from T0 to T1 the temperature reported by octoprint freezes to to last known temperature of T0.

#### Did the same happen when running OctoPrint in safe mode?

Doesn't matter.

#### Branch & Commit or Version of OctoPrint

1.3.4

#### Operating System running OctoPrint

OctoPI 0.13

#### Printer model & used firmware incl. version

Prusa I3 custom made by myself

#### Browser and Version of Browser, Operating System running Browser

Win 7 x64 with Chrome 60.0.x

#### Link to octoprint.log

provided in a screen capture above.

#### Link to contents of terminal tab or serial.log

provided in a screen capture above.

#### Link to contents of Javascript console in the browser

not captured.

#### Screenshot(s)/video(s) showing the problem:

provided in a screen capture above.

JohnOCFII, 2017-08-13:
Is that dark mode a plug-in?  You might want to try it without plug-ins active as my temperatures are displayed correctly.

cristiansavaro, 2017-08-15:
Just tried your theory with no luck. The same thing happens without that dark theme plugin.

foosel, 2017-08-21:
Please provide a full `serial.log` (you'll need to enable it) and `octoprint.log` (should always be written). Especially the `serial.log` - while you have provided a screenshot of your Terminal, it's scrolled up and naturally not showing the full history, which is of utmost importance to get to the ground of this.

cristiansavaro, 2017-08-21:
[image]

octoprint.log : 

`2017-08-21 15:58:39,917 - octoprint.server - INFO - ******************************************************************************
2017-08-21 15:58:39,918 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-08-21 15:58:39,918 - octoprint.server - INFO - ******************************************************************************
2017-08-21 15:58:39,977 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-08-21 15:58:42,020 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 15 mixin implementations, 13 hook handlers
2017-08-21 15:58:42,041 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-08-21 15:58:42,618 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 15 mixin implementations, 13 hook handlers
2017-08-21 15:58:42,678 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2017-08-21 15:58:42,680 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2017-08-21 15:58:42,828 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-08-21 15:58:44,873 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2017-08-21 15:58:46,933 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-08-21 15:58:46,934 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-08-21 15:58:46,937 - octoprint.plugin.core - INFO - Initialized 15 plugin implementation(s)
2017-08-21 15:58:46,946 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Cyborg Theme (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cyborgtheme
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
| !Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery
|  EEPROM Marlin Editor Plugin (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  Fullscreen Plugin (0.0.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Navbar Temperature Plugin (0.8-svc) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager
|  Print History Plugin (1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
|  Room Temperature Plugin (1.0-svc) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_roomtemp
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate
|  System Command Editor (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_systemcommandeditor
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/virtual_printer
2017-08-21 15:58:46,950 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-08-21 15:58:46,953 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-08-21 15:58:46,956 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2017-08-21 15:58:46,972 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2017-08-21 15:58:47,416 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2017-08-21 15:58:47,429 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-08-21 15:58:47,495 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-08-21 15:58:47,731 - octoprint.plugins.pluginmanager - INFO - Loaded notices from disk, was still valid
2017-08-21 15:58:47,875 - octoprint.plugins.softwareupdate - WARNING - Software update hook systemcommandeditor provides check for itself but that was already registered by eeprom_marlin - overwriting that third party registration now!
2017-08-21 15:58:47,929 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://xxxxxx:8080/'}
2017-08-21 15:58:48,355 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-08-21 15:58:48,378 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.192.254
[... cut ...]
2017-08-21 16:08:59,777 - Recv: ok T:58.24 /0.00 B:27.61 /0.00 @:0 B@:0
2017-08-21 16:09:04,769 - Send: M105
2017-08-21 16:09:04,777 - Recv: ok T:57.45 /0.00 B:27.50 /0.00 @:0 B@:0
2017-08-21 16:09:09,770 - Send: M105
2017-08-21 16:09:09,779 - Recv: ok T:56.83 /0.00 B:27.61 /0.00 @:0 B@:0
2017-08-21 16:09:14,771 - Send: M105
2017-08-21 16:09:14,780 - Recv: ok T:56.08 /0.00 B:27.64 /0.00 @:0 B@:0
2017-08-21 16:09:19,773 - Send: M105
2017-08-21 16:09:19,788 - Recv: ok T:55.49 /0.00 B:27.50 /0.00 @:0 B@:0
`

cristiansavaro, 2017-08-21:
By the way this is how I have octoprint configured in terms of printer profile : 

[image]

mattgaspar, 2017-09-04:
@foosel - I am also experiencing the same issue with shared nozzle enabled in OctoPrint.  I have Marlin setup as single nozzle and have a Y splitter above the hotend.  Printing works fine with OctoPrint but the temperature on T1 doesn't graph correctly.  I updated to the RC build of OctoPrint and have the latest version of Marlin (1.1.5).  I have been having this issue for a few month (when I first turned on single nozzle/switching extruder)

Version 1.3.5rc3 (rc/maintenance branch)

foosel, 2017-09-04:
I'm currently down sick and in no shape to touch code, but I'm going to take a look ASAP

galamdring, 2017-09-20:
I am seeing this issue as well. As best I can tell, it is due to the printer only reporting a T temperature. Octoprint is assigning that temperature to the current tool, but the temperature graph is not recording a T1 temp when T1 is selected. I'm going to look at this if you don't get to it first @foosel

foosel, 2017-09-20:
@galamdring I'm still down sick and help is always appreciated in any case, so please go ahead :)

galamdring, 2017-09-20:
From what I see, it is a question of either changing the logic that supplies the data to temperature.jinja2 so that when it does the foreach: tools it only gets one, or on the printer level, replicating the temp to each tool when sharedNozzle is enabled.

foosel, 2017-09-20:
Replication on the comm level might indeed be the answer here, if I understand the issue correctly.

galamdring, 2017-09-20:
I was able to correct the issue in master by replicating the one tool temp in octoprint.printer.standard._addTemperatureData to all of the extruders. When applying the same fix to maintenance, it does not work. The temperature delivered from comm isn't updating when tool 1 is selected.

cristiansavaro, 2017-09-21:
@galamdring can you provide a patch to test out your solution ?

galamdring, 2017-09-21:
@cristiansavaro are you running the master branch or maintenance? If you are running master, you could test with my fork. If you are running maintenance, I'm still working through that, tho I think I have a solution.

galamdring, 2017-09-21:
In Master, the single temperature being reported by the firmware was always being assigned to T0 in the comm layer, but not being reported correctly to the temp graph. Replicating the temperature across all tools corrected the issue.

Changes in the maintenance branch caused the single temp to be assigned to T(current_tool), but when that tool was not T0, the temps were not being assigned into the last_temperature dictionary that was being sent to the printer layer. 

Changed the check for if T0 in parsedTemps.keys to look for the current tool to confirm we had tool temps, then check if we have a sharednozzle, and replicating the temps across all tools from the current tool in _processTemperatures seems to have fixed the problem. Going to run through some non-sharedNozzle scenarios to confirm that it doesn't break anything, and I'll submit the pull request.

foosel, 2017-09-29:
Thanks to the PR by @galamdring this has now been fixed on `maintenance` (soon on `devel`) and will be part of the 1.3.6 release.

cristiansavaro, 2017-09-30:
thanks @foosel !

closed 2017-09-30 by cristiansavaro