OctoPrint/OctoPrint #2078: [Request] Hide webcam without loginning in

leobel96, 2017-08-14:
Hello, I would like to make my Octoprint installation visible also from outside Lan using haproxy, port forwarding and my public ip. Using the login system I will prevent the access to the system to other people. Is it possible to also hide webcam to people that haven't logged in? I don't like that other people can see what I am printing. Thank you!

RealTadango, 2017-08-15:
You'd better setup another way to prevent access to the entire system. They can also view what you are printing through the gcode viewer....

leobel96, 2017-08-15:
In fact I disabled G-CODE viewer :)

leobel96, 2017-08-15:
But isn't there a way to disable webcam stream without authentication?

leobel96, 2017-08-17:
Unfortunately I'm not a developer or an IT

malnvenshorn, 2017-08-18:
It's not very difficult to write own plugins for OctoPrint if you follow the documentation. If you decide to give it a try here a snippet how it could be accomplished to show the webcam only to logged in users (untested):

```
self.onStartup = function() {
    var element = $("#webcam_container");
    if (element.length) {
        element.attr("data-bind", element.attr("data-bind") + ", visible: loginState.isUser");
    }
};
self.onBeforeBinding = function() {
    self.control.loginState.isUser.subscribe(function() {
        if (self.control.loginState.isUser()) {
            self.control._enableWebcam();
       } else {
            self.control._disableWebcam();
        }
 });
 ```

Or use HTTP authentication as I mentioned before ;-)

leobel96, 2017-08-18:
I've also haproxy installed because of Octopi so I'll try with HTTP auth using it. Thank you all!

foosel, 2017-08-21:
* ["Why is there no access control for the webcam in OctoPrint" in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#why-is-there-no-access-control-for-the-webcam-in-octoprint)
  * [OctoPrint On Air #10 on how to configure HTTP authentication in haproxy](https://www.youtube.com/watch?v=dF5ndNxDH-M&t=2813s)

And apart from that this appears to be basically a duplicate of #1110 :)

closed 2017-08-21 by foosel