OctoPrint/OctoPrint #2082: Can't Update Octoprint--Octoprint Version==0+Unknown

mearleycf, 2017-08-17:
#### What were you doing?
1. I have installed octoprint on Windows 10 system
2. Octoprint serve is running in a virtualenv labeled oct
2.a I went into Software Update settings and set the location to c:\octoprint (I'm wondering if this is the problem since the app is actually running in a virtualenv)
3. I can access the GUI, and I am prompted to update now for my version to 1.3.4
4. Looking at my version information, it lists the octoprint version as 0+unknown

#### What did you expect to happen?
1. Octoprint to have a valid version number
2. Octoprint to update to 1.3.4

#### What happened instead?
4. When attempting to update, I get the following: 
-----*****
Updating, please wait.
+++++++++++++++++++++++++++++++
Now updating OctoPrint to 1.3.4
+++++++++++++++++++++++++++++++
c:\OctoPrint\oct\Scripts\python.exe "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py" --branch=master --force=false "c:\octoprint" 1.3.4
Exception in thread Thread-1:
Traceback (most recent call last):
File "C:\Python27\Lib\threading.py", line 810, in __bootstrap_inner
self.run()
File "C:\Python27\Lib\threading.py", line 763, in run
self.__target(*self.__args, **self.__kwargs)
File "build\bdist.win32\egg\sarge\__init__.py", line 1136, in run_node
result = getattr(self, method)(node, input, async)
File "build\bdist.win32\egg\sarge\__init__.py", line 1282, in run_command_node
node.cmd.run(input=input, async=async)
File "build\bdist.win32\egg\sarge\__init__.py", line 632, in run
self.process = p = Popen(self.args, **self.kwargs)
File "C:\Python27\Lib\subprocess.py", line 710, in __init__
errread, errwrite)
File "C:\Python27\Lib\subprocess.py", line 958, in _execute_child
startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
Error while trying to run command git.cmd diff --shortstat
Traceback (most recent call last):
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 313, in <module>
main()
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 309, in main
update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 204, in update_source
if _rescue_changes(git_executable, folder):
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 180, in _rescue_changes
raise RuntimeError("Could not update, \"git diff\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "git diff" failed with returncode None
Python executable: 'c:\\OctoPrint\\oct\\Scripts\\python.exe'
>>> Running: git diff --shortstat
> git.cmd diff --shortstat
The update did not finish successfully. Please consult the log for details.
-----*****

#### Did the same happen when running OctoPrint in safe mode?
Here's what happened when running in safe mode: 
-----*****
Updating, please wait.
+++++++++++++++++++++++++++++++
Now updating OctoPrint to 1.3.4
+++++++++++++++++++++++++++++++
c:\OctoPrint\oct\Scripts\python.exe "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py" --branch=master --force=false "c:\octoprint" 1.3.4
Exception in thread Thread-1:
Traceback (most recent call last):
File "C:\Python27\Lib\threading.py", line 810, in __bootstrap_inner
self.run()
File "C:\Python27\Lib\threading.py", line 763, in run
self.__target(*self.__args, **self.__kwargs)
File "build\bdist.win32\egg\sarge\__init__.py", line 1136, in run_node
result = getattr(self, method)(node, input, async)
File "build\bdist.win32\egg\sarge\__init__.py", line 1282, in run_command_node
node.cmd.run(input=input, async=async)
File "build\bdist.win32\egg\sarge\__init__.py", line 632, in run
self.process = p = Popen(self.args, **self.kwargs)
File "C:\Python27\Lib\subprocess.py", line 710, in __init__
errread, errwrite)
File "C:\Python27\Lib\subprocess.py", line 958, in _execute_child
startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
Error while trying to run command git.cmd diff --shortstat
Traceback (most recent call last):
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 313, in <module>
main()
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 309, in main
update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 204, in update_source
if _rescue_changes(git_executable, folder):
File "c:\OctoPrint\oct\lib\site-packages\octoprint-0+unknown-py2.7.egg\octoprint\plugins\softwareupdate\scripts\update-octoprint.py", line 180, in _rescue_changes
raise RuntimeError("Could not update, \"git diff\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "git diff" failed with returncode None
Python executable: 'c:\\OctoPrint\\oct\\Scripts\\python.exe'
>>> Running: git diff --shortstat
> git.cmd diff --shortstat
The update did not finish successfully. Please consult the log for details.
-----*****

#### Branch & Commit or Version of OctoPrint
0+unknown
c:\OctoPrint>git rev-parse HEAD
fe481e12b3c50a2def3b41515200e13a8c6a5c72

#### Operating System running OctoPrint

Windows version 10.0.14393

#### Printer model & used firmware incl. version

M3D Micro 1st gen

#### Browser and Version of Browser, Operating System running Browser
Version 60.0.3112.90 (Official Build) (64-bit)

#### Link to octoprint.log

https://gist.github.com/mearleycf/d155fbad07131b107c63835dfb15513a

#### Link to contents of terminal tab or serial.log

empty

#### Link to contents of Javascript console in the browser
--

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2017-08-21:
> Looking at my version information, it lists the octoprint version as 0+unknown

This sounds like a problem with your installation. Might be related to your issues, might not.

How exactly did you install this in the first place?

> `RuntimeError: Could not update, "git diff" failed with returncode None`

For some reason `git diff` is failing for your install and this returncode doesn't make it look like the command even executes properly. Which is weird.

What does `git --version` say?

ntoff, 2017-08-22:
Note: All of this info is from my own personal experience, I'm not a developer (unless you count being able to copy / paste) so take this info with a grain of salt.

According to this: https://stackoverflow.com/questions/8947140/git-cmd-vs-git-exe-what-is-the-difference-and-which-one-should-be-used

git.cmd is no longer included with github for windows and, at least in my case, this is true which causes octoprint's updater to fail as it specifically calls `git.cmd` rather than git.exe (even though apparently the updater script looks for either? I think? It was a couple of octoprint versions ago that I looked).

Even with adding git.cmd back, since windows locks files that are in use, "octoprint.exe" can't be updated while octoprint is running, and thus as far as I've been able to figure out, it can't be auto updated under windows. Yes the update will "work" for all the other files but you'll be left with an old octoprint.exe and thus the install will break.

I would love to be proven wrong but as it stands, I've never been able to update under windows without doing it manually through git pull then python clean / install.

Here is the contents of git.cmd https://github.com/msysgit/msysgit/blob/Git-1.7.9-preview20120201/cmd/git.cmd which you can save to your own git for windows install directory if you want to try it for yourself but as I said from my own experiences, windows' file locks prevent the auto updater from functioning correctly.

mearleycf, 2017-09-06:
git version 2.14.1.windows.1 from Command Prompt

foosel, 2017-10-02:
> How exactly did you install this in the first place?

closed 2017-10-19 by foosel