OctoPrint/OctoPrint #2083: Octoprint is scaling the rPi camera video stream

sundansx, 2017-08-18:
**What were you doing?**

viewing piCam in browser
**What did you expect to happen?**
I expected the video window to be displayed at 640x480 as specified in the prefs.

**various jpgs
** showing profile and printer settings.
What happened instead?

Video is being scaled to 588px × 441px.  cura video monitor shows the video in the correct resolution, albeit, upside down.
[image]

Version: 1.3.4 (master branch)
**Printer model & used firmware incl. version**

Qidi Tech 1, sailfish 7.8 (new printer)
**Browser and Version of Browser, Operating System running Browser**

Firefox 54.0.1 (64 bit), Win10
Same thing happens with Chrome Version 47.0.2526.106
**Link to octoprint.log**

included in attached zip file
**Link to contents of terminal tab or serial.log**

not printing yet..
**Link to contents of Javascript console in the browser**

NA?
**Screenshot(s) showing the problem:**
**Including screenshots and other files from above**
I have read the FAQ.

ntoff, 2017-08-20:
a 640px wide picture won't fit in the table size octoprint uses for the tab content and OctoPrint doesn't do any sort of dynamic content resizing (you could try the screen squish plugin). It's only the image you see in the web interface that gets resized, the timelapse image should be clear and at their normal resolution.

foosel, 2017-08-21:
As @ntoff already pointed out, there's no space for the full sized video feed in OctoPrint's control tab, so it scales to fit and there's no way around that (and now, just increasing the space there to allow the feed to get displayed at native resolution is no solution - people *are* in fact running full HD feeds...). So unless we are misunderstanding your problem here, this is working completely as intended.

sundansx, 2017-08-22:
Ok, I see what is going on.  Thanks for the explanation.  A feature I did not know about that I discovered during this investigation was the double click functionality.  That looks like it gives me a full rez image (scaled to a larger size, which should work fine).  I will close this issue.  Thanks for your help Gina and ntoff.  Gina - ease back in, don't want to see you get burned out again.  Glad you had a nice vacation.

closed 2017-08-22 by sundansx

foosel, 2017-08-22:
> the double click functionality

That's a [plugin](http://plugins.octoprint.org/plugins/fullscreen_webcam/) :D

> ease back in

❤️