OctoPrint/OctoPrint #2088: Save full video?

ahalekelly, 2017-08-23:
When trying to figure out why a print failed, the time lapse video goes by way too quickly. A full framerate video to look back at to figure out what went wrong would be very helpful, would it be possible to add a way to record the entire video off the camera and save it? I know it would take a lot of space, but you could set it to only hold, for example, the last 24 hours of video, video from the last couple prints, or start overwriting the oldest video when the SD card gets close to full?

foosel, 2017-08-23:
Someone could write a plugin that does that, yes. It really would fill up a hard disk quite quickly though, so that's a problem that would need to be solved first.

cilynx, 2018-12-03:
I've found that the best way to do this is to offload the heavy lifting to your client by recording the web stream.  For example, I use `mplayer http://orac/webcam/?action=stream -dumpstream -dumpfile output.mjpg` on my laptop to record machine runs.  If you're more of the GUI persuasion, you can [record web streams with VLC](https://confluence.bethel.edu/display/TLT/Recording+a+Network+Stream+with+VLC+Player) as well.  When using VLC, I've found that the Record button does not acknowledge being pressed, but the file is saved as it should be.