OctoPrint/OctoPrint #2092: Adding Print Option Popup

PowerShellCrack, 2017-08-23:
I curious if this has ever been recommended. I'm looking into developing a plugin that allows the user to change print settings right before print. Almost every time I print something, I am either changing the print temps from what the gcode listed and octoprint time-lapse. Then right after I have to go into print history plugin and change the filament spool using. Wouldn't it be nice to be able to change those (maybe more) right after you click the print button. A popup screen could come up like this: 
[https://goo.gl/photos/fishasnpCTD3Rajg8](url)

I created this popup message from paint to illustrate what I am looking at doing. I'm still in the very beginning stages of  development and would like some idea on the best way to do this.

FHeilmann, 2017-08-23:
At least with the print temp you can, as far as I know, use the offset box in the temperature tab.

Implementing what you suggest is tricky, because octoprint would have to traverse the whole file prior to printing to replace the temperature commands. And what if the slicer places multiple commands into the g-code (e.g. S3D allows to insert temperature calls at every layer).

I sort of agree with the timelapse and filament roll thing, but I believe there is a filament spool tracker plugin that you can use at least.

PowerShellCrack, 2017-08-23:
What if the plugin stated to not have the slice set the temps instead allow Octoprint to set them. I do that currently with Cura by going to the advanced option and commenting out the temp section. I then first preload the temp I want in Octoprint (base on my temp profiles) and start printing. I'm one of the guys that has a whole ton of gcode loading into octoprint and I don't use the same filament for them all. I also tend to change my mind on what filament I want to use based on my original idea for whatever reason. In the old days I would have to load it up in a slicer and make sure all the other in-fill, resolution and whatnot are correct, then set the temp based on what filament I want to use and save the gcode. Then I fined I missed something and do it all over. 

OR
Parsing the file gcode wouldn't be to difficult; I can do it in most Windows scripting languages (powershell, cscript, batch), I'm sure Linux/python/perl has some capabilities of that. Heck maybe there could be an advanced upload button that does it when uploaded (if plugin is installed and that feature is enabled)
[https://stackoverflow.com/questions/21653738/parsing-large-9gb-file-using-python](url)

Just ideas. It would be a nice feature to speed up the printing process and reduce mistakes.

ntoff, 2017-08-24:
>What if the plugin stated as a prereq to not have the slicer set the temps because instead it will have Octoprint set them.

Then you've eliminated the possibility of ever using simplify3d's multiple processes, or people with multiple nozzles that might have one set for support material, or even a single nozzle that is set to print support material cooler so it is easier to break away. Varying the temperature based on whether or not the slicer can slow down enough for small features goes back as far as skeinforge, all of those nice features people have grown used to would be rendered useless.

>Parsing a large gcode file wouldn't be too difficult;

Don't forget people are running octoprint on pi zero's now, aren't they like 1 core @ 1ghz or something? Running off a (relatively) slow SD card. What might take under a  second on a PC might take quite a while on a pi zero. You'll want to make sure you postpone scanning if there's a print job in progress in case it causes performance issues.

closed 2017-09-07 by GitIssueBot