OctoPrint/OctoPrint #2093: Naming for Feed rate and Flow rate

sanchosk, 2017-08-24:
#### What were you doing?

I am trying to slow-down a print in the middle of the process. I am not sure if I should choose feed or flow rate - as a non-native person, both English terms make the same sense for me in part.

#### What did you expect to happen?

I expect one of the sliders to be renamed to Speed factor and the other to Extrusion factor to better make sense.
Keep it in line with [http://reprap.org/wiki/G-code#M220:_Set_speed_factor_override_percentage](RepRap Wiki) if possible.

#### What happened instead?

The names are often confusing for me and I end up sending direct G-code instead.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (rc/devel branch) 

#### Operating System running OctoPrint

Linux

#### Printer model & used firmware incl. version

Prusa i3 clone

#### Browser and Version of Browser, Operating System running Browser

All browsers

#### Link to octoprint.log

Not relevant.

#### Link to contents of terminal tab or serial.log

Not relevant.

#### Link to contents of Javascript console in the browser

Not relevant.

#### Screenshot(s)/video(s) showing the problem:

Not relevant.

RealTadango, 2017-08-25:
If feed rate was called motor speed is was more clear. That basically is what it is: The speed of all motors (including the filament feed). 

But for the naming sake, it matches the printer firmware names but i don't like them either.

nophead, 2017-08-25:
Feed rate is common CNC machine terminology from long before CNC machines had extruders. It does become a bit ambiguous when you add an extruder that you feed filament through.

foosel, 2017-08-25:
What @nophead said. I'd rather not diverge from established terminology if possible. I could be talked into a tool tip though. Alternatively, someone could write a plugin that simply replaces the text on these buttons in the front-end.

Note that "Feed rate" is located in the X/Y/Z section and "Flow rate" in the "Tool (E)" section for a reason as well.

sanchosk, 2017-08-25:
That "established terminology" was relevant for CNC maybe. Even G-code wiki
uses different terms. So does every other 3D printing software.
I believe the decision to solve this by plugin is short-sighted.
Vast majority of user base does not use the OctoPRINT for CNC, but for
printing - therefore I'd suggest the opposite - rename the items properly
in OctoPrint for everyone aligned with G-code naming and introduce a plugin
for CNC userbase.

foosel, 2017-08-25:
The 3d printing community has been ignoring and sometimes even outright abusing established terminology, processes and solutions from CNC since the beginning. "G-CODE" for example - no, what we are used to in 3D printing only vaguely resembles the actual standardized G-CODE. Which is what got us into a whole mess with regards to communication "protocols" among other things which I could rant about for hours, and also keeps leading to discussions like this.

If I now change this wording to reflect what's on that Wiki page (which btw EVERYONE can just edit at any time - it isn't a standards document sadly and cannot be trusted blindly, even though it is the closest to a standard that we have), you can bet on another ticket being opened within hours of releasing an adjusted version by a person using OctoPrint with their printer but used to the CNC terminology, complaining about the change and demanding to return to the old naming. Or someone who's used to the current terminology from their printer's own firmware/controller panel, which also states feedrate/flowrate. For reference on how firmware handles this:
  * Marlin: feedrate & flowrate
    * http://marlinfw.org/docs/gcode/M220.html and http://marlinfw.org/docs/gcode/M221.html
    * https://github.com/MarlinFirmware/Marlin/blob/d3bc877007ceaeaeed11898998ee83b54f83de46/Marlin/Marlin_main.cpp#L168-L169
  * Repetier: feedrate & flowrate
    * https://github.com/repetier/Repetier-Firmware/blob/9ac153211cf94797fdad4381087686ebaa2f3e91/src/ArduinoAVR/Repetier/Repetier.ino#L115-L116
  * Smoothie: speed factor override & flowrate (hooray, a mixture of both)
    * http://smoothieware.org/supported-g-codes

Doing a change like this always leads to disruption, so it has to be thought through well and demand has to be checked. 

I do not want to add configuration options for stuff like this since "just add an option" leads to config pages no one can navigate anymore (and that's already bad enough in OctoPrint as-is). The plugin system is in place to - among other things - also solve cases of personal taste and work flows.

sanchosk, 2017-08-25:
That's funny. When I send M220 S100 to Repetier FW, it replies back with SpeedMultiply:100.
For M221 Repetier refers to Extrusion flow multiplier - again, much more distinguishable then just flow.

But as the whole 3D printing community and CNC community using OctoPrint so far is happy with this and nobody found it to be confusing, I will simply do the change on my side within the code directly - if I am the only one non-native speaker that has trouble to distinguish the terms, I can for sure live with such solution.

closed 2017-08-25 by sanchosk

foosel, 2017-08-25:
I'm also not a native speaker and don't have a "proper" CNC background. I just remind myself that the only thing that flows is molten plastic, and that the stepper movements themselves are therefore the other thing ¯\_(ツ)_/¯

RichyT, 2019-03-14:
@foosel, you say "Note that "Feed rate" is located in the X/Y/Z section and "Flow rate" in the "Tool (E)" section for a reason as well." but it is clearly stated elsewhere  that the Feed rate also influences extrusion speed.

This is not the first time I have seen someone clarify what appears (I think?) to be correct then someone else drops a confusion bomb into the discussion.

As you may guess, I'm here because I was looking at a feed button and a flow button and it's *NOT* clear how they interact. I'm not going to reopen this but I hope people will take a look at this and find a way to make it more logical. Yes, even a tool-top will help. In case anyone is so inclined, talking to me as if I'm an idiot won't help and others will keep asking the same question.